betrusted-io/xous-core #350: Vault app: U2F works with DUO, but does not show up on FIDO list

kop316, 2023-03-29:
DUO Mobile works with the Vault app, but the vault app does not seem to know it is being registered with it, and does not show up.

This is probably a low priority since I imagine the issue is with DUO, but I figured it would be good to document.

spoelstraethan, 2023-07-28:
I ran into the same thing with GitHub where I had previously registered a Mooltipass and a couple YubiKeys, it seems if the device you are adding isn't the FIRST key in, GitHub may send a U2F request rather than a FIDO2/WebAuthN request, and then Precursor gets set up with it, but the FIDO view isn't set to show that type of credential in its filter.

spoelstraethan, 2023-09-01:
I've gotten a little more insight into this. It appears U2F and FIDO2 are supported and show up in the FIDO tab of the interface, BUT I _think_ that maybe FIDO which don't supply a relying party or identity hint only show up in `pddb keylist opensk`.  I was adding some entries for GitHub and Gmail where I already had a Yubikey security key present, and that appears to cause the site to return a different registration type, if I removed all the other keys or selected Passkey for Google, then it got registered as a FIDO2.

Not sure if this is related, but it  wasn't until I had a secret basis mounted that I started seeing my new FIDO entries, when it was only .System they showed up in the pddb dictlist and pddb keylist opensk but I wasn't seeing anything in the interface. I need to try wiping the device and re-registering with different sites after mounting an initial secret basis to see whether those "missing" entries are directly attributable to code that isn't showing FIDO entries from .System properly.