betrusted-io/xous-core #361: gdb-stub instability

bunnie, 2023-04-08:
Alright, got things upgraded and tried running it. It sort of works, but the connection seems unstable. This is what I get on a "good" run:

```
pi@betrusted-dev:~/code/betrusted-scripts $ riscv-none-elf-gdb -ex 'tar ext /dev/ttyS0' -ex 'attach 27' vault
GNU gdb (xPack GNU RISC-V Embedded GCC armv7l) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=armv7l-unknown-linux-gnueabihf --target=riscv-none-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
vault: No such file or directory.
Remote debugging using /dev/ttyS0
Ignoring packet error, continuing...
warning: unrecognized item "
> q
> S
> u
> p
Printing processes
Process 1 state: Running(100)  TID: 0  Memory mapping: (satp: 0x804409a0, mode: 1, ASID: 1, PPN: 409a0000) conns:0/32 kernel
Process 2 state: Sleeping  TID: 2  Memory mapping: (satp: 0x80840a56, mode: 1, ASID: 2, PPN: 40a56000) conns:5/32 xous-ticktimer
Process 3 state: Sleeping  TID: 3  Memory mapping: (satp: 0x80c40a4f, mode: 1, ASID: 3, PPN: 40a4f000) conns:1/32 xous-log
Process 4 state: Sleeping  TID: 2  Memory mapping: (satp: 0x81040a48, mode: 1, ASID: 4, PPN: 40a48000) conns:2/32 xous-names
Process 5 state: Sleeping  TID: 0  Memory mapping: (satp: 0x81440a41, mode: 1, ASID: 5, PPN: 40a41000) conns:21/32 xous-susres
Process 6 state: Sleeping  TID: 2  Memory mapping: (satp: 0x81840a3a, mode: 1, ASID: 6, PPN: 40a3a000) conns:7/32 graphics-server
Process 7 state: Sleeping  TID: 2  Memory mapping: (satp: 0x81c40a33, mode: 1, ASID: 7, PPN: 40a33000) conns:10/32 keyboard
Process 8 state: Sleeping  TID: 2  Memory mapping: (satp: 0x82040a2c, mode: 1, ASID: 8, PPN: 40a2c000) conns:8/32 spinor
Process 9 state: Sleeping  TID: 2  Memory mapping: (satp: 0x82440a25, mode: 1, ASID: 9, PPN: 40a25000) conns:10/32 llio
Process 10 state: Sleeping  TID: 2  Memory mapping: (satp: 0x82840a1e, mode: 1, ASID: 10, PPN: 40a1e000) conns:9/32 com
Process 11 state: Sleeping  TID: 4  Memory mapping: (satp: 0x82c40a17, mode: 1, ASID: 11, PPN: 40a17000) conns:17/32 dns
Process 12 state: Sleeping  TID: 2  Memory mapping: (satp: 0x83040a10, mode: 1, ASID: 12, PPN: 40a10000) conns:21/32 gam
Process 13 state: Sleeping  TID: 2  Memory mapping: (satp: 0x83440a09, mode: 1, ASID: 13, PPN: 40a09000) conns:7/32 ime-frontend
Process 14 state: Sleeping  TID: 2  Memory mapping: (satp: 0x83840a02, mode: 1, ASID: 14, PPN: 40a02000) conns:8/32 codec
Process 15 state: Sleeping  TID: 2  Memory mapping: (satp: 0x83c409fb, mode: 1, ASID: 15, PPN: 409fb000) conns:6/32 modals
Process 16 state: Sleeping  TID: 2  Memory mapping: (satp: 0x840409f4, mode: 1, ASID: 16, PPN: 409f4000) conns:16/32 root-keys
Process 17 state: Sleeping  TID: 2  Memory mapping: (satp: 0x844409ed, mode: 1, ASID: 17, PPN: 409ed000) conns:5/32 trng
Process 18 state: Sleeping  TID: 2  Memory mapping: (satp: 0x848409e6, mode: 1, ASID: 18, PPN: 409e6000) conns:6/32 sha2
Process 19 state: Sleeping  TID: 3  Memory mapping: (satp: 0x84c409df, mode: 1, ASID: 19, PPN: 409df000) conns:6/32 engine-25519
Process 20 state: Sleeping  TID: 2  Memory mapping: (satp: 0x850409d8, mode: 1, ASID: 20, PPN: 409d8000) conns:5/32 jtag
Process 21 state: Sleeping  TID: 2  Memory mapping: (satp: 0x854409d1, mode: 1, ASID: 21, PPN: 409d1000) conns:29/32 status
Process 22 state: Sleeping  TID: 5  Memory mapping: (satp: 0x858409ca, mode: 1, ASID: 22, PPN: 409ca000) conns:16/32 pddb
Process 23 state: Sleeping  TID: 2  Memory mapping: (satp: 0x85c409c3, mode: 1, ASID: 23, PPN: 409c3000) conns:10/32 usb-device-xous
Process 24 state: Sleeping  TID: 14  Memory mapping: (satp: 0x860409bc, mode: 1, ASID: 24, PPN: 409bc000) conns:17/32 net
Process 25 state: Sleeping  TID: 2  Memory mapping: (satp: 0x864409b5, mode: 1, ASID: 25, PPN: 409b5000) conns:3/32 ime-plugin-shell
Process 26 state: Sleeping  TID: 2  Memory mapping: (satp: 0x868409ae, mode: 1, ASID: 26, PPN: 409ae000) conns:18/32 shellchat
Process 27 state: Sleeping  TID: 2  Memory mapping: (satp: 0x86c409a7, mode: 1, ASID: 27, PPN: 409a7000) conns:18/32 vault
" in "qSupported" response
^CQuit
Don't know how to attach.  Try "help target".
(gdb) quit
```

Of course in the above trial I had the path of the executable wrong, but at least it did something. More typically, I get something like this:

```
pi@betrusted-dev:~/code/betrusted-scripts $ riscv-none-elf-gdb -ex 'tar ext /dev/ttyS0' -ex 'attach 27' ~/vault
GNU gdb (xPack GNU RISC-V Embedded GCC armv7l) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=armv7l-unknown-linux-gnueabihf --target=riscv-none-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/vault...
Remote debugging using /dev/ttyS0
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
Don't know how to attach.  Try "help target".
(gdb)
```

I wonder if there are flow control issues possibly? I guess there are no RTS/CTS pins on this interface, but basically if the kernel is held up you're going to lose data. The FIFO is only 16 bytes deep on the Tx/Rx queues on the UARTs, so if you write more than that and the kernel is busy with something else, you'll end up dropping bytes.

_Originally posted by @bunnie in https://github.com/betrusted-io/xous-core/issues/360#issuecomment-1500917353_

bunnie, 2023-04-08:
I've moved this to an issue so it's more visible for tracking, since the PR comments disappear as soon as they are closed.

fwiw, anyone trying to reproduce this will need to know a trick, you need to type `console app` into shellchat to switch the UART mux to the GDB UART port on the hardware for this all to work.

xobs, 2023-04-08:
"Printing processes" cannot be printed out the GDB serial port. If you're seeing that, then you have the wrong port muxed.

Isn't there a command to switch the mux so that the GDB port (which uses the "APP_UART" controller) goes out the serial port?

bunnie, 2023-04-08:
it should be switched over to the app_uart port. hmm...let me poke around more.

bunnie, 2023-04-08:
ah, i think i found a configuration problem. i'll keep poking at it.

bunnie, 2023-04-08:
alright, it helps to actually have the app_uart instantiated. i had stripped it out because it was causing trouble in timing closure and stability in the gateware, figuring I could add it back in later if and when this feature was added.

Looks like it's attaching now, but still unstable:

```
pi@betrusted-dev:~/code/betrusted-scripts $ riscv-none-elf-gdb -ex 'tar ext /dev/ttyS0' -ex 'attach 27' ~/vault
GNU gdb (xPack GNU RISC-V Embedded GCC armv7l) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=armv7l-unknown-linux-gnueabihf --target=riscv-none-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/vault...
Remote debugging using /dev/ttyS0
Attaching to program: /home/pi/vault, process 27
[New Thread 27.3]
[New Thread 27.4]
[New Thread 27.5]
[New Thread 27.6]
[New Thread 27.7]
[New Thread 27.8]
[New Thread 27.9]
[New Thread 27.10]
[New Thread 27.11]
0x0011d520 in xous::syscall::receive_message ()
(gdb) info thr
  Id   Target Id         Frame
* 1    Thread 27.2       0x0011d520 in xous::syscall::receive_message ()
  2    Thread 27.3       0x0008417e in vault::actions::start_actions_thread::{{closure}} ()
  3    Thread 27.4       0x0011d520 in xous::syscall::receive_message ()
  4    Thread 27.5       0x0011c26e in xous::syscall::send_message ()
  5    Thread 27.6       0x00053c80 in core::ops::function::FnOnce::call_once{{vtable.shim}} ()
  6    Thread 27.7       0x000529cc in core::ops::function::FnOnce::call_once{{vtable.shim}} ()
  7    Thread 27.8       Ignoring packet error, continuing...
stepi
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

xobs, 2023-04-09:
This does seem like a flow control issue, since it completely died in the middle of the transfer. I bet if you did "set debug remote 1" before connecting, you'd see erroneous transactions.

I'll have to think about this...

xobs, 2023-04-09:
Can I have you do two things?

1. Try to run with `set debug remote 1`. I'm sure it'll just reflect the fact that we're running into buffering issues, but I'm curious just the same.
2. Try applying the following patch, which stuffs a ring buffer into the receive path. This might relieve some pressure on the FIFOs which appear to be 16 bytes in hardware:

```patch
diff --git a/kernel/src/debug/gdb.rs b/kernel/src/debug/gdb.rs
index c39ec920..59543a3a 100644
--- a/kernel/src/debug/gdb.rs
+++ b/kernel/src/debug/gdb.rs
@@ -49,15 +49,78 @@ impl ProcessPid for XousTarget {
     }
 }
 
+struct MicroRingBuf<const N: usize = 32> {
+    buffer: [u8; N],
+    head: usize,
+    tail: usize,
+}
+
+impl<const N: usize> Default for MicroRingBuf<N> {
+    fn default() -> Self {
+        MicroRingBuf {
+            buffer: [0u8; N],
+            head: 0,
+            tail: 0,
+        }
+    }
+}
+
+impl<const N: usize> MicroRingBuf<N> {
+    // pub fn capacity(&self) -> usize {
+    //     self.buffer.len()
+    // }
+    // pub fn len(&self) -> usize {
+    //     self.head.wrapping_sub(self.tail) % N
+    // }
+
+    pub fn is_full(&self) -> bool {
+        (self.tail.wrapping_sub(1) % N) == self.head
+    }
+
+    pub fn try_push(&mut self, val: u8) -> Result<(), ()> {
+        if self.is_full() {
+            return Err(());
+        }
+        self.buffer[self.head] = val;
+        self.head = (self.head + 1) % N;
+        Ok(())
+    }
+
+    pub fn try_pop(&mut self) -> Option<u8> {
+        if self.tail == self.head {
+            return None;
+        }
+        let val = self.buffer[self.tail];
+        self.tail = (self.tail + 1) % N;
+        Some(val)
+    }
+}
+
 fn receive_irq(uart: &mut GdbUart) {
-    while let Some(c) = uart.getc() {
+    let mut buffer = MicroRingBuf::<32>::default();
+    loop {
+        // Try to fill up the ring buffer with as many characters
+        // as can fit. This is to compensate for the fact that we do
+        // all of this processing in an interrupt context, and the
+        // hardware UART buffer is only a few characters deep.
+        while !buffer.is_full() {
+            if let Some(c) = uart.getc() {
+                buffer.try_push(c).ok();
+            } else {
+                break;
+            }
+        }
+
+        // If there is a character in the buffer, process it. Otherwise,
+        // we're done.
+        let Some(c) = buffer.try_pop() else { break };
         process_character(c);
-    }
 
-    // If the GDB server goes away for some reason, reconstitute it
-    unsafe {
-        if GDB_STATE.is_none() {
-            init();
+        // If the GDB server goes away for some reason, reconstitute it
+        unsafe {
+            if GDB_STATE.is_none() {
+                init();
+            }
         }
     }
 }
```

bunnie, 2023-04-09:
What do you want me to do with debug remote? here's an example with `stepi`. It seems to be much more stable with debug remote turned on, I haven't seen it crash yet. Maybe the debugging information introduces enough delay that it doesn't trigger an overflow.

```
Remote debugging using /dev/ttyS0
Attaching to program: /home/pi/vault, process 27
[New Thread 27.3]
[New Thread 27.4]
[New Thread 27.5]
[New Thread 27.6]
[New Thread 27.7]
[New Thread 27.8]
[New Thread 27.9]
[New Thread 27.10]
[New Thread 27.11]
0x0011d530 in xous::syscall::receive_message ()
(gdb) set debug remote 1
(gdb) stepi
[remote] Sending packet: $m11d4d4,2#5d
[remote] Packet received: 97f0
[remote] Sending packet: $m11d4d6,2#5f
[remote] Packet received: ffff
[remote] Sending packet: $m11d4d8,2#61
[remote] Packet received: e780
[remote] Sending packet: $m11d4da,2#8a
[remote] Packet received: 20ba
[remote] Sending packet: $vCont?#49
[remote] Packet received: vCont;c;C;s;S
[remote] packet_ok: Packet vCont (verbose-resume) is supported
[remote] Sending packet: $vCont;s:p1b.2;c:p1b.-1#bc
[remote] wait: enter
[remote] wait: exit

^C[remote] pass_ctrlc: enter
  [remote] interrupt: enter
  [remote] interrupt: exit
[remote] pass_ctrlc: exit
[remote] wait: enter
  [remote] Packet received: S02
  [remote] select_thread_for_ambiguous_stop_reply: enter
    [remote] select_thread_for_ambiguous_stop_reply: process_wide_stop = 0
    [remote] select_thread_for_ambiguous_stop_reply: first resumed thread is Thread 27.2
    [remote] select_thread_for_ambiguous_stop_reply: is this guess ambiguous? = 1
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
  [remote] select_thread_for_ambiguous_stop_reply: exit
[remote] wait: exit
[remote] Sending packet: $g#67
[remote] Packet received: 000000001cd5110040edff7f0000000000000060100000003cea11006cedff7f18f3ff7f4affff7f0f00000061113753027ced79e2492122cf363078000000000000000000000000b8000020c80000200600000004000000030000000000000098f0ff7f01000000800000206800002050edff7ff40000008fffffff0300000030d51100
[remote] Sending packet: $qfThreadInfo#bb
[remote] Packet received: mp1b.02,p1b.03,p1b.04,p1b.05,p1b.06,p1b.07,p1b.08,p1b.09,p1b.0a,p1b.0b
[remote] Sending packet: $qsThreadInfo#c8
[remote] Packet received: l

Thread 1 received signal SIGINT, Interrupt.
0x0011d530 in xous::syscall::receive_message ()
```

xobs, 2023-04-09:
I could believe that.

What if you leave `debug remote` off and add the little mini buffer patch I posted?

xobs, 2023-04-09:
Also, previously it crashed when you did `info thr`. Does it crash with `info thr` with debugging turned on?

bunnie, 2023-04-09:
hm, maybe it's `info thr` that does it. here's the output with debugging turned on, and the patch applied:

```
(gdb) info thr
[remote] Sending packet: $qfThreadInfo#bb
[remote] Packet received: mp1b.02,p1b.03,p1b.04,p1b.05,p1b.06,p1b.07,p1b.08,p1b.09,p1b.0a,p1b.0b
[remote] Sending packet: $qsThreadInfo#c8
[remote] Packet received: l
[remote] Sending packet: $qThreadExtraInfo,p1b.2#e8
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000002#7a
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000003#7b
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000004#7c
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000005#7d
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000006#7e
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000007#7f
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000008#80
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000009#81
[remote] Packet received:
[remote] Sending packet: $qP0000001f000000000000000a#a9
[remote] Packet received:
[remote] Sending packet: $qP0000001f000000000000000b#aa
[remote] Packet received:
  Id   Target Id         Frame
[remote] Sending packet: $qP0000001f0000000000000002#7a
[remote] Packet received:
* 1    Thread 27.2       0x0011d530 in xous::syscall::receive_message ()
[remote] Sending packet: $qP0000001f0000000000000003#7b
[remote] Packet received:
[remote] Sending packet: $Hgp1b.3#13
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Packet received: 0000000070bd0c00100002600000000000500260580000003cea110080020260030000000f0000000f000000ef08df6bc79d5c1f3cff8313ad686f7d00000000000000000000000004000000ffffff7f090000004100000003000000cc020260ac11010001000000c0020260a03f026030021200030000000000000000000000aac10c00
  2    Thread 27.3       0x000cc1aa in <p256::arithmetic::projective::ProjectivePoint as core::ops::arith::Mul<&p256::arithmetic::scalar::Scalar>>::mul ()
[remote] Sending packet: $qP0000001f0000000000000004#7c
[remote] Packet received:
[remote] Sending packet: $Hgp1b.4#14
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Packet received: 000000001cd51100307d04600000000000900460100000003cea11005c7d0460587e0460050000000f000000b0b5471cd0bc46ee433114c668cc1e0d000000000000000000000000f07e046001000000ffffff7f807f04600000000001c014001802002070000020680000200f0000000000000000000000000000000000000030d51100
  3    Thread 27.4       0x0011d530 in xous::syscall::receive_message ()
[remote] Sending packet: $qP0000001f0000000000000005#7d
[remote] Packet received:
[remote] Sending packet: $Hgp1b.5#15
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Packet received: 000000006ac2110040a306600000000000c00660001000003cea110058b0146008ae0660010000001000000013000000010000000b00000000b01460001000000000000000100000001000008cb502008cac0660010000000200000000b014600400000000b014608800000088ac0660bffffffff40000008fffffff030000007ec21100
  4    Thread 27.5       0x0011c27e in xous::syscall::send_message ()
[remote] Sending packet: $qP0000001f0000000000000006#7e
[remote] Packet received:
[remote] Sending packet: $Hgp1b.6#16
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Packet received: 0000000038bc090050d608600000000000f0086094df0860000000002800096003000000000009600f0000006fe7ae919f3464858164f961c38782dd00000000000000000000000008d7086010db08600100000018000000e7030000000000000010000014d708600000000004000000bffffffff40000008fffffff0300000014bd0900
  5    Thread 27.6       0x0009bd14 in cbor::reader::Reader::decode_complete_data_item ()
[remote] Sending packet: $qP0000001f0000000000000007#7f
[remote] Packet received:
[remote] Sending packet: $Hgp1b.7#17
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Packet received: 00000000b4a90900403e0b600000000000500b60100000003cea1100c43d0b6065f8f044900400200f000000ee0e558365f8f044b62289e046bd41ee000000000000000000000000843f0b609ca00200c800002046bd41ee680000200100000002000000b62289e0ee0e5583b83e0b600000000000000000000000000000000060aa0900
  6    Thread 27.7       0x0009aa60 in base32::decode ()
[remote] Sending packet: $qP0000001f0000000000000008#80
[remote] Packet received:
[remote] Sending packet: $Hgp1b.8#18
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
  7    Thread 27.8       Ignoring packet error, continuing...
[remote] Sending packet: $g#67
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
Ignoring packet error, continuing...
[remote] send_g_packet: Bad register packet; fetching a new packet
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
Ignoring packet error, continuing...
[remote] send_g_packet: Bad register packet; fetching a new packet
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
Ignoring packet error, continuing...
[remote] send_g_packet: Bad register packet; fetching a new packet
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
Ignoring packet error, continuing...
[remote] send_g_packet: Bad register packet; fetching a new packet
[remote] getpkt_or_notif_sane_1: Timed out.
```

note that `step` and `stepi` just hang forever. Even though gdb doesn't report an error, the system doesn't seem to single step. I would expect the `vault` app to be hung until the `step` is done, but I'm able to interact with the app so maybe it's not actually single stepping?

xobs, 2023-04-09:
It might be there's a bug you're running into that I didn't catch.

How can I reproduce this? When I run `cargo xtask app-image vault` it complains that there is no `app_uart` in the svd file.

bunnie, 2023-04-09:
you need to be at commit rev f3dc258a6a5a31998c3c9941427eff256c216afd or later in xous-core to reproduce. Best if you just fetch from head. utralib, as you may recall, is now hosted on crates.io so you have to redo all the version pins to fix that. HEAD currently has that fixed.

There's some other bug that's happening right now that is causing factory test to fail, but I think it may be unrelated to your patch and just "serendipitously" appearing around the same time. Still figuring it out.

bunnie, 2023-04-09:
Ah no. It does look like it could be related to this. I'm getting this now on resume from suspend after adding your patch:

```
!!! Unrecognized exception: UserSoftwareInterrupt(0)
PROGRAM HALT: CPU Exception on PID 5: User swi from 0x00000000
Current thread 2:
Thread 2:
PC:ffd0e93c   SP:fff7ff80   RA:ffd0e958
GP:00000000   TP:00000000
T0:0000000a   T1:ffd0ed06   T2:00026000
T3:00000001   T4:ffd80058   T5:00001400   T6:ffc00514
S0:00000005   S1:ffd82358   S2:00000001   S3:0000003c
S4:ffd81458   S5:ffcd0000   S6:00000040   S7:00000006
S8:0000004c   S9:00000fff  S10:40ff4c88  S11:000009e4
A0:00000007   A1:00000005   A2:00000000   A3:00000000
A4:00000000   A5:00000000   A6:00000000   A7:00000000
Memory Maps for PID 5:
    1021 Superpage for ff400000 @ 40735000 (flags: VALID)
        1021 ff7fd000 -> 40735000 (flags: VALID | R | W | A | D)
        1022 ff7fe000 -> 40736000 (flags: VALID | R | W | A | D)
        1023 ff7ff000 -> 40732000 (flags: VALID | R | W | A | D)
    1022 Superpage for ff800000 @ 40736000 (flags: VALID)
           0 ff800000 -> 40737000 (flags: VALID | R | W | A | D)
           1 ff801000 -> 40734000 (flags: VALID | R | W | A | D)
    1023 Superpage for ffc00000 @ 40732000 (flags: VALID)
        <unavailable>
End of map
[!] Terminating process with PID 5
PANIC in PID 1: panicked at 'couldn't switch to pid: InvalidSyscall', kernel/src/main.rs:114:22
```

This looks like it may be related to the code you touched, if you have any ideas on what could be causing this I'm all ears

bunnie, 2023-04-09:
(This is on a kernel that is not built with gdb-stub, as is the current default for releases)

xobs, 2023-04-09:
If gdb crashes, then it gets restarted immediately. As part of the termination process, it resumes any process that was suspended.

One thing that I've noticed is that thread 8 (which GDB confusingly calls TID 7) has code at a different offset than thread 5 and 6:

```
(gdb) info thr
  Id   Target Id         Frame
* 1    Thread 27.2       0x0011bf8a in xous::syscall::send_message ()
  2    Thread 27.3       0x00083db8 in vault::actions::start_actions_thread::{{closure}} ()
  3    Thread 27.4       0x0011d2e6 in xous::syscall::receive_message ()
  4    Thread 27.5       0x0011bf8a in xous::syscall::send_message ()
  5    Thread 27.6       0x0005372e in core::ops::function::FnOnce::call_once{{vtable.shim}} ()
  6    Thread 27.7       0x0005247a in core::ops::function::FnOnce::call_once{{vtable.shim}} ()
  7    Thread 27.8       0x000c2ef6 in core::ops::function::FnOnce::call_once{{vtable.shim}} ()
  8    Thread 27.9       0x000c180e in core::ops::function::FnOnce::call_once{{vtable.shim}} ()
  9    Thread 27.10      0x0011bf8a in xous::syscall::send_message ()
  10   Thread 27.11      0x0011d2e6 in xous::syscall::receive_message ()
(gdb)
```

Might it be that those addresses are alighed oddly?

xobs, 2023-04-09:
Well, the good news is that I can reproduce the crash in Renode. Let me work on that.

xobs, 2023-04-09:
dbf28fbd61abca5d555fa595008845e9a1961dc0 fixes the resume issue

bunnie, 2023-04-09:
thanks for fixing that!

xobs, 2023-04-09:
Do you still get the crash if you add the ringbuffer patch I posted?

bunnie, 2023-04-09:
the most recent gdb trace i posted here is with your patch.

xobs, 2023-04-09:
This is bizarre. And it's unfortuante that we can't see what's happening on the kernel serial port, because I think the error message would appear there.

This in particular doesn't make sense:

```
[remote] Sending packet: $Hgp1b.7#17
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Packet received: 00000000b4a90900403e0b600000000000500b60100000003cea1100c43d0b6065f8f044900400200f000000ee0e558365f8f044b62289e046bd41ee000000000000000000000000843f0b609ca00200c800002046bd41ee680000200100000002000000b62289e0ee0e5583b83e0b600000000000000000000000000000000060aa0900
  6    Thread 27.7       0x0009aa60 in base32::decode ()
[remote] Sending packet: $qP0000001f0000000000000008#80
[remote] Packet received:
[remote] Sending packet: $Hgp1b.8#18
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
  7    Thread 27.8       Ignoring packet error, continuing...
```

The `$H` packet tells GDB to specify a thread for subsequent operations. Specifically, it's setting `g` for "other operations", and "p" (process) 0x1b.0x08 (i.e. process 27, thread 8). But there's no response.

Contrast that with what we get under emulation:

```
[remote] Sending packet: $Hgp1b.7#17
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Packet received: 00000000aa260500401e0b600000000000300b60100000003aeb1100c41d0b601f08df35180400200f0000009090f9511f08df352c7e5875d02f5f07000000000000000000000000841f0b6024610100c8000020d02f5f076800002001000000020000002c7e58759090f951b81e0b600000000000000000000000000000000056270500
  7    Thread 27.7       0x00052756 in core::ops::function::FnOnce::call_once{{vtable.shim}} ()
[remote] Sending packet: $qP0000001f0000000000000008#80
[remote] Packet received:
[remote] Sending packet: $Hgp1b.8#18
[remote] Packet received: OK
```

For some reason, under your setup it just stops responding during that query.

bunnie, 2023-04-09:
if the query wasn't well-formed (e.g. truncated somehow) how would the system respond?

bunnie, 2023-04-09:
In particular, this packet:
`[remote] Sending packet: $qP0000001f0000000000000008#80`

is longer than the Rx FIFO?

xobs, 2023-04-09:
I was thinking that it might be the case that `$Hgp1b.8#18` is getting corrupted, yeah. Maybe the stub is missing characters?

Can I have you run `set remote noack-packet off` prior to connecting to the remote? I.e.:

```
set remote noack-packet off
tar ext /dev/ttyS0
att 27
info thr
```

Note that the `disconnect` command doesn't appear to work entirely, as gdb doesn't always send the disconnect packet, so it may be necessary to reboot the target. We definitely need to improve protocol recovery.

bunnie, 2023-04-09:
here's what i'm getting:

```
$ riscv-none-elf-gdb -ex 'tar ext /dev/ttyS0' -ex 'attach 27' ~/vault
GNU gdb (xPack GNU RISC-V Embedded GCC armv7l) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=armv7l-unknown-linux-gnueabihf --target=riscv-none-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/vault...
Remote debugging using /dev/ttyS0
Attaching to program: /home/pi/vault, process 27
[New Thread 27.3]
[New Thread 27.4]
[New Thread 27.5]
[New Thread 27.6]
[New Thread 27.7]
[New Thread 27.8]
[New Thread 27.9]
[New Thread 27.10]
[New Thread 27.11]
0x0011d530 in xous::syscall::receive_message ()
(gdb) set remote noack-packet off
(gdb) set debug remote 1
(gdb) info thr
[remote] Sending packet: $qfThreadInfo#bb
[remote] Packet received: mp1b.02,p1b.03,p1b.04,p1b.05,p1b.06,p1b.07,p1b.08,p1b.09,p1b.0a,p1b.0b
[remote] Sending packet: $qsThreadInfo#c8
[remote] Packet received: l
[remote] Sending packet: $qThreadExtraInfo,p1b.2#e8
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000002#7a
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000003#7b
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000004#7c
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000005#7d
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000006#7e
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000007#7f
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000008#80
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000009#81
[remote] Packet received:
[remote] Sending packet: $qP0000001f000000000000000a#a9
[remote] Packet received:
[remote] Sending packet: $qP0000001f000000000000000b#aa
[remote] Packet received:
  Id   Target Id         Frame
[remote] Sending packet: $qP0000001f0000000000000002#7a
[remote] Packet received:
* 1    Thread 27.2       0x0011d530 in xous::syscall::receive_message ()
[remote] Sending packet: $qP0000001f0000000000000003#7b
[remote] Packet received:
[remote] Sending packet: $Hgp1b.3#13
[remote] Packet received: OK
[remote] Sending packet: $g#67
[... 31 lines cut ...]
[remote] Packet received:
[remote] Sending packet: $Hgp1b.8#18
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Packet received: 000000005244060010ad0d600000000000c00d6000c20d601c1b0f000000000064af0d60400500200f000000062e07b7ecbb8aa0fe94bd4964264aa5000000000000000000000000fe94bd49062e07b79121000000001200a43b020064264aa5ecbb8aa00300000001000000090000000000000000000000000000000000000090450600
  7    Thread 27.8       0x00064590 in vault::ctap::CtapState::process_fido_command ()
[remote] Sending packet: $qP0000001f0000000000000009#81
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
  8    Ignoring packet error, continuing...
[remote] Sending packet: $Hgp1b.9#19
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
Thread 27.9       Ignoring packet error, continuing...
[remote] Sending packet: $g#67
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
Ignoring packet error, continuing...
[remote] send_g_packet: Bad register packet; fetching a new packet
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
```

bunnie, 2023-04-09:
looks like it got further at least?

xobs, 2023-04-09:
That looks the same. You need to `set remote noack-packet off` before you connect, but in that log you've set it after already connecting.

Can you run the following:

```
riscv-none-elf-gdb -ex 'set remote noack-packet off' -ex 'set debug remote 1' -ex 'tar ext /dev/ttyS0' -ex 'attach 27' ~/vault
```

bunnie, 2023-04-09:
Ah. I did not know that you could not set that after connecting. It seems to work better now:
```
pi@betrusted-dev:~/code/betrusted-scripts $ riscv-none-elf-gdb -ex 'set remote noack-packet off' -ex 'set debug remote 1' -ex 'tar ext /dev/ttyS0' -ex 'attach 27' ~/vault
GNU gdb (xPack GNU RISC-V Embedded GCC armv7l) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=armv7l-unknown-linux-gnueabihf --target=riscv-none-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/vault...
Remote debugging using /dev/ttyS0
[remote] start_remote_1: enter
  [remote] Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+#ec
  [remote] Received Ack
  [remote] Packet received: PacketSize=1000;vContSupported+;multiprocess+;QStartNoAckMode+;qXfer:features:read+
  [remote] packet_ok: Packet qSupported (supported-packets) is supported
  [remote] Sending packet: $vMustReplyEmpty#3a
  [remote] Received Ack
  [remote] Packet received:
  [remote] Sending packet: $!#21
  [remote] Received Ack
  [remote] Packet received: OK
  [remote] Sending packet: $Hgp0.0#ad
  [remote] Received Ack
  [remote] Packet received: E01
  [remote] Sending packet: $qXfer:features:read:target.xml:0,ffb#79
  [remote] Received Ack
  [remote] Packet received: m<target version="1.0"><architecture>riscv:rv32</architecture></target>
  [remote] Sending packet: $qXfer:features:read:target.xml:46,ffb#b3
  [remote] Received Ack
  [remote] Packet received: l
  [remote] Sending packet: $qTStatus#49
  [remote] Received Ack
  [remote] Packet received:
  [remote] packet_ok: Packet qTStatus (trace-status) is NOT supported
  [remote] Sending packet: $?#3f
  [remote] Received Ack
  [remote] Packet received: W00;
[remote] start_remote_1: exit
Attaching to program: /home/pi/vault, process 27
[remote] Sending packet: $vAttach;1b#99
[remote] Received Ack
[remote] Packet received: T05thread:p1b.02;
[remote] packet_ok: Packet vAttach (attach) is supported
[remote] Sending packet: $qC#b4
[remote] Received Ack
[remote] Packet received: QCp1b.02
[remote] Sending packet: $Hgp1b.2#12
[remote] Received Ack
[remote] Packet received: OK
[remote] Sending packet: $qXfer:features:read:target.xml:0,ffb#79
[remote] Received Ack
[remote] Packet received: m<target version="1.0"><architecture>riscv:rv32</architecture></target>
[remote] Sending packet: $qXfer:features:read:target.xml:46,ffb#b3
[remote] Received Ack
[remote] Packet received: l
[remote] wait: enter
[remote] wait: exit
[remote] Sending packet: $g#67
[remote] Received Ack
[remote] Packet received: 000000001cd5110040edff7f0000000000000060100000003cea11006cedff7f18f3ff7f4affff7f0f000000f828609c343df59a1a67de7c86b5a44f000000000000000000000000b8000020c80000200600000004000000030000000000000098f0ff7f01000000800000206800002050edff7ff40000008fffffff0300000030d51100
[... 135 lines cut ...]
[remote] Received Ack
[remote] Packet received: 000000004c2d060060d90f600000000000f00f6000f20f601c1b0f000000000005000000801000200f0000001dc3389bc9f122badd073f70ec1a7ae6000000000000000000000000e80500200000000000000000c0d90f6000000000c9f122ba00000000000000001dc3389bdd073f7000000000000000000000000000000000122e0600
  8    Thread 27.9       0x00062e12 in vault::ctap::CtapState::process_fido_command ()
[remote] Sending packet: $qP0000001f000000000000000a#a9
[remote] Received Ack
[remote] Packet received:
[remote] Sending packet: $Hgp1b.a#41
[remote] Received Ack
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Received Ack
[remote] Packet received: 000000006ac21100a00e126000000000002012600000000000000000b40e1260580f1260170000000e000000000000001b000000000000000a0000000000000000000000ec23d8ff03000000380600200c0000007c0f1260040000000100000060040000040000001000000009000000000000000000000000000000000000007ec21100
  9    Thread 27.10      0x0011c27e in xous::syscall::send_message ()
[remote] Sending packet: $qP0000001f000000000000000b#aa
[remote] Received Ack
[remote] Packet received:
[remote] Sending packet: $Hgp1b.b#42
[remote] Received Ack
[remote] Packet received: OK
[remote] Sending packet: $g#67
[remote] Received Ack
[remote] Packet received: 000000001cd51100f03c14600000000000801460100000003cea11001c3d1460183e1460050000000f000000f1b9b2544d930869363d8bb0921354dd000000000000000000000000f83f146005000000000000001000000001000000b06f1460184401005845010000000000020000000000000000000000000000000000000030d51100
  10   Thread 27.11      0x0011d530 in xous::syscall::receive_message ()
(gdb)
```

xobs, 2023-04-09:
Non-stop mode appears to be useful for transports that are error-free. Perhaps we do need to support non-stop mode by default, in which case I can look into disabling support for that.

If you do `set remote debug 0`, can you debug things normally?

bunnie, 2023-04-09:
although, trying more stuff seems to eventually fail:
```
(gdb) stepi
[remote] Sending packet: $Hgp1b.2#12
[remote] Received Ack
[remote] Packet received: OK
[remote] Sending packet: $m11d4d4,2#5d
[remote] Received Ack
[remote] Packet received: 97f0
[remote] Sending packet: $m11d4d6,2#5f
[remote] Received Ack
[remote] Packet received: ffff
[remote] Sending packet: $m11d4d8,2#61
[remote] Received Ack
[remote] Packet received: e780
[remote] Sending packet: $m11d4da,2#8a
[remote] Received Ack
[remote] Packet received: 20ba
[remote] Sending packet: $vCont?#49
[remote] Received Ack
[remote] Packet received: vCont;c;C;s;S
[remote] packet_ok: Packet vCont (verbose-resume) is supported
[remote] Sending packet: $vCont;s:p1b.2;c:p1b.-1#bc
[remote] Received Ack
[remote] wait: enter
[remote] wait: exit

^C[remote] pass_ctrlc: enter
  [remote] interrupt: enter
  [remote] interrupt: exit
[remote] pass_ctrlc: exit
[remote] wait: enter
  [remote] Packet received: S02
  [remote] select_thread_for_ambiguous_stop_reply: enter
    [remote] select_thread_for_ambiguous_stop_reply: process_wide_stop = 0
    [remote] select_thread_for_ambiguous_stop_reply: first resumed thread is Thread 27.2
    [remote] select_thread_for_ambiguous_stop_reply: is this guess ambiguous? = 1
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
  [remote] select_thread_for_ambiguous_stop_reply: exit
[remote] wait: exit
[remote] Sending packet: $g#67
[remote] Received Ack
[remote] Packet received: 000000001cd5110040edff7f0000000000000060100000003cea11006cedff7f18f3ff7f480a00200f000000f828609c343df59a1a67de7c86b5a44f000000000000000000000000b8000020c80000200600000004000000100a00200300000002000000010000008000002068000020bffffffff40000008fffffff0300000030d51100
[remote] Sending packet: $qfThreadInfo#bb
[remote] Received Ack
[remote] Packet received: mp1b.02,p1b.03,p1b.04,p1b.05,p1b.06,p1b.07,p1b.08,p1b.09,p1b.0b,p1b.0c,p1b.0d
[remote] Sending packet: $qsThreadInfo#c8
[remote] Received Ack
[remote] Packet received: l
[New Thread 27.12]
[New Thread 27.13]

Thread 1 received signal SIGINT, Interrupt.
0x0011d530 in xous::syscall::receive_message ()
(gdb) disassemble
Dump of assembler code for function _ZN4xous7syscall15receive_message17hba4890ef3ae81111E:
[remote] Sending packet: $m11d4c0,40#8a
[remote] Sending packet: $m11d4c0,40#8a
[remote] Sending packet: $m11d4c0,40#8a
[remote] Sending packet: $m11d4c0,40#8a
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
[remote] getpkt_or_notif_sane_1: Timed out.
   0x0011d4d4 <+0>:     Ignoring packet error, continuing...
terminate called after throwing an instance of 'gdb_exception_error'

Fatal signal: Aborted
----- Backtrace -----
0x524693 ???
0x617f2f ???
0x767fc1bf ???
        ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.

This is a bug, please report it.  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>.

Aborted
```

bunnie, 2023-04-09:
> Non-stop mode appears to be useful for transports that are error-free. Perhaps we do need to support non-stop mode by default, in which case I can look into disabling support for that.
> 
> If you do `set remote debug 0`, can you debug things normally?

No, it seems to not work at all:
```
pi@betrusted-dev:~/code/betrusted-scripts $ riscv-none-elf-gdb -ex 'set remote noack-packet off' -ex 'tar ext /dev/ttyS0' -ex 'attach 27' ~/vault
GNU gdb (xPack GNU RISC-V Embedded GCC armv7l) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=armv7l-unknown-linux-gnueabihf --target=riscv-none-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/vault...
Remote debugging using /dev/ttyS0
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
```

xobs, 2023-04-09:
That seems like what happens when gdbstub thinks gdb is still connected, but the connection is reset. In particular, that happens when the stub fails due to a protocol error, which is what happened  Give it another shot?

It looks like what actually happened there is GDB issued a `stepi` instruction which we patched, but for whatever reason the CPU never actually hit that instruction and never broke into the function.

So you hit Control-C and tried to manually break into it. But by then the stub had crashed. So when GDB tried to read memory from `0x11d4c0` it timed out and things fell over.

gdb is very prone to segfaulting or panicking

Overall it seems as though issues stem from the transport being unreliable, likely due to trying to cram everything into the interrupt handler. 

Perhaps it's a bug in the GdbUart handler, and I'm just not understanding the documentation correctly?

https://github.com/betrusted-io/xous-core/blob/89ca18caaddaa99d36fe6dc2d5c2aea4e2046ce0/kernel/src/platform/precursor/gdbuart.rs#L145-L152

Maybe it should be:

```
 if self.uart_csr.r(utra::app_uart::RXEMPTY) {
     return None;
 }

 let ret = Some(self.uart_csr.r(utra::app_uart::RXTX) as u8);
 self.uart_csr.wfo(utra::app_uart::EV_PENDING_RX, 1);
 Some(ret)
 } 
```

xobs, 2023-04-09:
What's the correct way to drain a lot of data from the UART and/or to acknowledge that data has been read? The doc at https://ci.betrusted.io/betrusted-soc/doc/uart.html doesn't really say how to use it...

bunnie, 2023-04-09:
> What's the correct way to drain a lot of data from the UART and/or to acknowledge that data has been read? The doc at https://ci.betrusted.io/betrusted-soc/doc/uart.html doesn't really say how to use it...

That's ancient autodoc stuff from some of the oldest code in LiteX :-/

The best I can tell is you would want to poll the RX_EMPTY/TX_EMPTY bits until they are showing a '1'. The _EMPTY bit series was added in a later patch, presumably explicitly for this purpose.

There is no further docs than that, unfortunately, but the _EMPTY bits are directly wired to the FIFO `empty` bits.

xobs, 2023-04-09:
Do you see any change if you make the patch that I posted above to `SerialRead`?

bunnie, 2023-04-09:
I didn't realize that was something you wanted me to try out. I'll try it soon.

bunnie, 2023-04-09:
Seems to be much more stable...i invoked GDB with this command line:

`riscv-none-elf-gdb -ex 'set remote noack-packet off' -ex 'tar ext /dev/ttyS0' -ex 'attach 27' ~/vault`

And I was able to get thread info, read registers, run code, stop running code, inspect registers, and there were no errors.

However, single stepping does not work, and breakpoints don't seem to work either. Basically I can run code and then ctrl-C and it'll break at some random point and I can see the line that is running and dump registers there. But I think breakpoints don't work?

fwiw the patch as posted doesn't work, I had to fix a couple minor things. Did you want me to push that patch and the other one up to main so we're not drifting too far from the mainline?

xobs, 2023-04-09:
Would you mind reworking the patch?

Does it still work if you don't select `noack-packet off`? It may just be that the code injection isn't working right.

bunnie, 2023-04-09:
> Would you mind reworking the patch?
> 
> Does it still work if you don't select `noack-packet off`? It may just be that the code injection isn't working right.

Not sure what you mean by reworking the patch, but in order to test it I had to make it work. The question I had was if it would be OK to push the current change set up to `main` so that it's in CI.

Yes, it seems stable now without `noack-packet off`.

xobs, 2023-04-09:
Yes, please push the current change set up to `main`

Just to be clear -- this isn't an XIP binary, is it? It seems as though updating program memory isn't working for some reason. Or maybe breakpoints aren't implemented on the CPU?

bunnie, 2023-04-09:
oh! right. vault is `xip` on this binary.

bunnie, 2023-04-09:
i've loaded a non-xip version of `vault`, and now instead of not breaking, the system hangs. This is what I get:

```
(gdb) info reg
ra             0x11d51c 0x11d51c <xous::syscall::receive_message+56>
sp             0x7fffed40       0x7fffed40
gp             0x0      0x0
tp             0x60000000       0x60000000
t0             0x10     16
t1             0x11ea3c 1174076
t2             0x7fffed6c       2147478892
fp             0x7ffff318       0x7ffff318
s1             0x200009e8       536873448
a0             0xf      15
a1             0xbaf15991       -1158588015
a2             0x2d03351a       755184922
a3             0x3fa51678       1067783800
a4             0x13e22d1f       333589791
a5             0x0      0
a6             0x0      0
a7             0x0      0
s2             0x200000b8       536871096
s3             0x200000c8       536871112
s4             0x6      6
s5             0x4      4
s6             0x200009b0       536873392
s7             0x3      3
s8             0x2      2
s9             0x1      1
s10            0x20000080       536871040
s11            0x20000068       536871016
t3             0xffffffbf       -65
t4             0xf4     244
t5             0xffffff8f       -113
t6             0x3      3
pc             0x11d530 0x11d530 <xous::syscall::receive_message+76>
(gdb) stepi
[remote] Sending packet: $Hgp1b.2#12
[remote] Received Ack
[remote] Packet received: OK
[remote] Sending packet: $m11d4e4,2#5e
[remote] Received Ack
[remote] Packet received: 3571
[remote] Sending packet: $m11d4e6,2#60
[remote] Received Ack
[remote] Packet received: 06cf
[remote] Sending packet: $m11d4e8,2#62
[remote] Received Ack
[remote] Packet received: 22cd
[remote] Sending packet: $m11d4ea,2#8b
[remote] Received Ack
[remote] Packet received: 2a84
[remote] Sending packet: $m11d4ec,2#8d
[remote] Received Ack
[remote] Packet received: 2800
[remote] Sending packet: $m11d4ee,2#8f
[remote] Received Ack
[remote] Packet received: 4146
[remote] Sending packet: $vCont?#49
[remote] Received Ack
[remote] Packet received: vCont;c;C;s;S
[remote] packet_ok: Packet vCont (verbose-resume) is supported
[remote] Sending packet: $vCont;s:p1b.2;c:p1b.-1#bc
[remote] Received Ack
[remote] wait: enter
[remote] wait: exit
```

The connection seems stable at least when I'm not trying to step or break (I can do `info thr`, `info reg`, etc.)

xobs, 2023-04-09:
I'm happy to add support for hardware breakpoints and watchpoints if we have processor support for them. But for now breakpoints and single-stepping only work on non-XIP programs.

It looks like you're still running in ACK mode. That's fine, but it will probably run slightly faster if you remove the `noack` request prior to connecting.

Regardless, it should at least work. Here's what you should see:

```
(gdb) stepi
[remote] Sending packet: $m11c012,2#23
[remote] Received Ack
[remote] Packet received: 7571
[remote] Sending packet: $m11c014,2#25
[remote] Received Ack
[remote] Packet received: 06c7
[remote] Sending packet: $m11c016,2#27
[remote] Received Ack
[remote] Packet received: 22c5
[remote] Sending packet: $m11c018,2#29
[remote] Received Ack
[remote] Packet received: 2a84
[remote] Sending packet: $m11c01a,2#52
[remote] Received Ack
[remote] Packet received: 4945
[remote] Sending packet: $vCont?#49
[remote] Received Ack
[remote] Packet received: vCont;c;C;s;S
[remote] packet_ok: Packet vCont (verbose-resume) is supported
[remote] Sending packet: $vCont;s:p1b.2;c:p1b.-1#bc
[remote] Received Ack
[remote] wait: enter
  [remote] Packet received: T96thread:p1b.02;
[remote] wait: exit
[remote] Sending packet: $g#67
[remote] Received Ack
[remote] Packet received: 00000000f8d2110040edff7f0000000000000060100000003aeb11006cedff7f18f3ff7f4affff7f090000000000161b030000000c00000000000000000000000000000000000000b8000020c80000200600000004000000030000000000000098f0ff7f010000008000002068000020bffffffff40000008fffffff030000000ed31100
[remote] Sending packet: $qfThreadInfo#bb
[remote] Received Ack
[remote] Packet received: mp1b.02,p1b.03,p1b.04,p1b.05,p1b.06,p1b.07,p1b.08,p1b.09,p1b.0a,p1b.0b
[remote] Sending packet: $qsThreadInfo#c8
[remote] Received Ack
[remote] Packet received: l

Thread 2 received signal EXC_BREAKPOINT, Breakpoint.
0x0011d30e in xous::syscall::receive_message ()
(gdb)
```

Given it's not actually breaking, I'm wondering if the CPU doesn't have ebreak support enabled. I do see this, which I'm assuming means it's enabled?

https://github.com/betrusted-io/pythondata-cpu-vexriscv/blob/dd8e0bb6c8e88936a2960c628cd54d45d7bcfe96/pythondata_cpu_vexriscv/verilog/src/main/scala/vexriscv/GenBetrustedSoC.scala#L125

I don't see anything inhibiting it, and I do think I see the compressed instruction decompressor at:

https://github.com/SpinalHDL/VexRiscv/blob/b29eb542f278edb8ee2c91863a5b26e76b5932c9/src/main/scala/vexriscv/plugin/Misc.scala#L98

Similarly, I think the actual decoder is at:

https://github.com/SpinalHDL/VexRiscv/blob/b29eb542f278edb8ee2c91863a5b26e76b5932c9/src/main/scala/vexriscv/plugin/CsrPlugin.scala#L846-L862

If neither breakpoints nor single-stepping (which use breakpoints) are working, then the vexriscv implementation is doing something terribly unexpected, and it'd be a good idea to figure out what's going on there.

xobs, 2023-04-09:
Just to be clear -- when you say "the system hangs" -- what do you mean? Do you mean it freezes and the watchdog timer hits to reboot it? Or does the program in question simply stop? What happens if you hit "Control-C" to break into the program being debugged?

bunnie, 2023-04-09:
Here's a bit more flavor on it, on a simpler server:
```
pi@betrusted-dev:~/code/betrusted-scripts $ riscv-none-elf-gdb -ex 'tar ext /dev/ttyS0' -ex 'attach 17' ~/trng
GNU gdb (xPack GNU RISC-V Embedded GCC armv7l) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=armv7l-unknown-linux-gnueabihf --target=riscv-none-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/pi/trng...
(No debugging symbols found in /home/pi/trng)
Remote debugging using /dev/ttyS0
Attaching to program: /home/pi/trng, process 17
[New Thread 17.3]
0x0002aeaa in xous::syscall::receive_message ()
(gdb) info thr
  Id   Target Id         Frame
* 1    Thread 17.2       0x0002aeaa in xous::syscall::receive_message ()
  2    Thread 17.3       0x0002beec in xous_api_susres::suspend_cb_server ()
(gdb) set debug remote 1
(gdb) info thr
[remote] Sending packet: $qfThreadInfo#bb
[remote] Packet received: mp11.02,p11.03
[remote] Sending packet: $qsThreadInfo#c8
[remote] Packet received: l
[remote] Sending packet: $qP0000001f0000000000000002#7a
[remote] Packet received:
[remote] Sending packet: $qP0000001f0000000000000003#7b
[remote] Packet received:
  Id   Target Id         Frame
[remote] Sending packet: $qP0000001f0000000000000002#7a
[remote] Packet received:
* 1    Thread 17.2       0x0002aeaa in xous::syscall::receive_message ()
[remote] Sending packet: $qP0000001f0000000000000003#7b
[remote] Packet received:
  2    Thread 17.3       0x0002beec in xous_api_susres::suspend_cb_server ()
```
At this point, the system is running and I'm able to interact with it on the keyboard. Now I try this:
```
(gdb) stepi
[remote] Sending packet: $Hgp11.2#e1
[remote] Packet received: OK
[remote] Sending packet: $m2ae5e,2#5d
[remote] Packet received: 3571
[remote] Sending packet: $m2ae60,2#29
[remote] Packet received: 06cf
[remote] Sending packet: $m2ae62,2#2b
[remote] Packet received: 22cd
[remote] Sending packet: $m2ae64,2#2d
[remote] Packet received: 2a84
[remote] Sending packet: $m2ae66,2#2f
[remote] Packet received: 2800
[remote] Sending packet: $m2ae68,2#31
[remote] Packet received: 4146
[remote] Sending packet: $vCont?#49
[remote] Packet received: vCont;c;C;s;S
[remote] packet_ok: Packet vCont (verbose-resume) is supported
[remote] Sending packet: $vCont;s:p11.2;c:p11.-1#5a
[remote] wait: enter
[remote] wait: exit
c
^C[remote] pass_ctrlc: enter
  [remote] interrupt: enter
  [remote] interrupt: exit
[remote] pass_ctrlc: exit
^C[remote] pass_ctrlc: enter
The target is not responding to interrupt requests.
Stop debugging it? (y or n) y
[remote] pass_ctrlc: exit
Disconnected from target.
(gdb) quit
```

At this point, the system is actually "hung" as in the status thread has stopped updating and no keyboard response. Eventually, the WDT hits and everything reboots.

I *think* what might actually be happening is the TRNG thread is stopped and this may be causing some other servers to get clogged waiting for a response from the TRNG and eventually the system reboots.

bunnie, 2023-04-09:
> If neither breakpoints nor single-stepping (which use breakpoints) are working, then the vexriscv implementation is doing something terribly unexpected, and it'd be a good idea to figure out what's going on there.

My first thought is it's a caching bug of some sort...this would be mutable code on the I-cache path and even though a flush is being called I wouldn't be surprised if there wasn't a bug of some sort there. Probably not many have tried this approach with the Vex before.

Maybe you could run a verilator of the CPU just to make sure the `ebreak` instruction is going through? I don't understand what `ebreak` does myself so I wouldn't be able to write a test case.

xobs, 2023-04-09:
`ebreak` is effectively another form of `ecall` -- it generates an exception which causes the exception handler to be called.

The behaviour you're describing sounds an awful lot like the kernel is panicking, possibly due to a write violation, but it also could be due to an unhandled exception.

This is where the patching occurs:

https://github.com/betrusted-io/xous-core/blob/cb875cf54eabe71558ebd95feb2826d36283913f/kernel/src/debug/gdb/riscv.rs#L112-L119

This gets written as four `u8` values, in order to avoid alignment issues:

https://github.com/betrusted-io/xous-core/blob/cb875cf54eabe71558ebd95feb2826d36283913f/kernel/src/debug/gdb/multi_thread_base.rs#L148-L154

The actual business end of the poke is here:

https://github.com/betrusted-io/xous-core/blob/cb875cf54eabe71558ebd95feb2826d36283913f/kernel/src/arch/riscv/mem.rs#L505-L557

Nothing in ther should panic, so that's rather odd.

bunnie, 2023-04-09:
Is there any way to get KP's to pop up a guru meditation? We have a perfectly serviceable screen that can be used to output panic information...

xobs, 2023-04-10:
I can't think of a way to get KPs to pop up on the screen. If the kernel is panicking, then things have gone exceedingly wrong. Not even scheduling is guaranteed to work, and you can't know who owns what memory.

Additionally, KPs generate a lot of data, including memory maps for the process that crashed. I don't think that would fit on the screen.

The best we could do is write it to a memory buffer somewhere and pull it out on the next reboot, except RAM is wiped during allocation.

bunnie, 2023-04-10:
Well, the KP doesn't have to be "safe"...it can leave the system state in ruins since the only next step is a reboot from there. Most KPs that happen not as a result of say, a loader error, are fairly self contained from what I've seen. 

I was thinking something like a base of what's used by the loader to show the messages on boot: https://github.com/betrusted-io/xous-core/blob/cb875cf54eabe71558ebd95feb2826d36283913f/loader/src/secboot.rs#L34-L198 it's compact and has no dependencies. The only trick would be creating a page table entry to the graphics frame buffer, and to the graphics CSR. But those are in known physical locations -- and the device-side mapping doesn't have to be exclusive, I think we could actually just map that at boot time probably, no? That is we can allocate a couple virtual pages that both point to the same physical pages without problems (of course you have concurrency problems, but if there's a KP, there's no more concurrency because the schedule has stopped).

xobs, 2023-04-10:
You mean, something like this?

[image]

xobs, 2023-04-10:
I've created a PR at https://github.com/betrusted-io/xous-core/pull/363 that adds this.

Would you mind incorporating that, then making the following temporary patch and seeing if there's anything useful on the screen?

```patch
diff --git a/kernel/src/arch/riscv/irq.rs b/kernel/src/arch/riscv/irq.rs
index 3aaa4742..10944816 100644
--- a/kernel/src/arch/riscv/irq.rs
+++ b/kernel/src/arch/riscv/irq.rs
@@ -355,7 +355,7 @@ pub extern "C" fn trap_handler(
         }

         _ => {
-            println!("!!! Unrecognized exception: {:?}", ex);
+            panic!("!!! Unrecognized RISC-V exception: {:?}", ex);
         }
     }

@@ -395,7 +395,7 @@ pub extern "C" fn trap_handler(
     // For now, let's halt the whole system instead so that it becomes
     // immediately obvious that we screwed up. On harware this will trigger
     // a watchdog reset.
-    println!(
+    panic!(
         "{}: CPU Exception on PID {}: {}",
         if is_kernel_failure {
             "!!! KERNEL FAILURE !!!"
```

bunnie, 2023-04-10:
> You mean, something like this?
> 
> [image]

nice! yah exactly like that. I think the top of the stack dump is all you need anyways to get a sense of what's going on. the pages and pages of page tables is less important.

bunnie, 2023-04-10:
I have incorporated the patches like this:

```rust
        _ => {
            #[cfg(not(any(feature = "precursor", feature = "renode")))]
            println!("!!! Unrecognized exception: {:?}", ex);
            #[cfg(any(feature = "precursor", feature = "renode"))]
            panic!("!!! Unrecognized exception: {:?}", ex);
        }
    }
```
and 
```rust
    // The exception was not handled. We should terminate the program here.
    // For now, let's halt the whole system instead so that it becomes
    // immediately obvious that we screwed up. On hardware this will trigger
    // a watchdog reset.
    #[cfg(not(any(feature = "precursor", feature = "renode")))]
    println!(
        "{}: CPU Exception on PID {}: {}",
        if is_kernel_failure {
            "!!! KERNEL FAILURE !!!"
        } else {
            "PROGRAM HALT"
        },
        pid,
        ex
    );
    #[cfg(any(feature = "precursor", feature = "renode"))]
    panic!(
        "{}: CPU Exception on PID {}: {}",
        if is_kernel_failure {
            "!!! KERNEL FAILURE !!!"
        } else {
            "PROGRAM HALT"
        },
        pid,
        ex
    );
```
And repeated the experiment where I `stepi` on the `trng` server.

The moment I run `trng pump` (which tickles the trng server) the UI freezes. There is no panic. Eventually, the watchdog timer triggers without a panic on the screen.

I'll also run another test with a deliberate panic just to make sure I did the patch right.

bunnie, 2023-04-10:
[image]
Panic printing works, at least!

bunnie, 2023-04-10:
> 

No, the TRNG is not running, I think, as the system instantly hangs including threads that shouldn't be dependent on the TRNG?

```
0x0002aeaa in xous::syscall::receive_message ()
(gdb) info thr
  Id   Target Id         Frame
* 1    Thread 17.2       0x0002aeaa in xous::syscall::receive_message ()
  2    Thread 17.3       0x0002beec in xous_api_susres::suspend_cb_server ()
(gdb) stepi
^Cc
^CThe target is not responding to interrupt requests.
Stop debugging it? (y or n) y
Disconnected from target.
(gdb) quit
```
It does not seem to resume.

xobs, 2023-04-10:
Hmm... One possibility is that it is a CPU bug where it's not properly delegating the execution.

What should happen is the program hits the `ebreak` instruction and generates Breakpoint exception. From my understanding, this should be taken in Machine Mode, except we delegate the exception to Supervisor mode.

Perhaps that delegation isn't happening...

bunnie, 2023-04-10:
Yes, I feel like this around the point where we break out the verilog simulator. 

The question is how to write the test case. I can build a kernel that has gdb in it, but, actually connecting GDB to a verilog simulator is going to be annoying. Is there some way I can maybe just trigger this as part of some sham code that runs early on in the execution flow?

xobs, 2023-04-10:
Yes, that appears to be the case -- debug exceptions simply aren't able to be delegated on vexriscv:

https://github.com/betrusted-io/VexRiscv/blob/51b69a1527c01616f386fa5cffb993313bfec919/src/main/scala/vexriscv/plugin/CsrPlugin.scala#L957-L960

Bit 3 is the debug exception, but you can see it goes 0, 1, 2, 4, 5, ...

So what happens is the program issues `ebreak`, which kicks the processor all the way back to Machine mode, and gets stuck here:

https://github.com/betrusted-io/xous-core/blob/b471b604d0eed7d67d8694b69bede8582a1262f6/loader/src/asm.rs#L51-L62

I've opened an issue here to see what it would take to get it added to the CPU, because this would be nice to get fixed so we can remove a source of DoS: https://github.com/SpinalHDL/VexRiscv/issues/328

[21 more replies]

closed 2023-12-11 by bunnie