betrusted-io/xous-core #373: `graphics-server` crashes on macos

xobs, 2023-04-27:
Since #371, macos now crashes in `minifb`, which is now thread 2.

This is likely because some operating systems including macos don't allow drawing to the screen from a thread other than TID 1. As a result, the graphics library is throwing an error which Rust fails to unwind properly.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               graphics-server [89829]
Path:                  /Users/USER/*/graphics-server
Identifier:            graphics-server
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        xous-kernel [89773]
Responsible:           Terminal [855]
User ID:               501

Date/Time:             2023-04-27 13:01:09.0077 +0700
OS Version:            macOS 13.3.1 (22E261)
Report Version:        12
Anonymous UUID:        78F39E87-2E56-B558-45A0-EF5910EFC0DF

Sleep/Wake UUID:       3838C284-A045-4B33-A4D4-9FDF2078B1C8

Time Awake Since Boot: 230000 seconds
Time Since Wake:       2482 seconds

System Integrity Protection: enabled

Crashed Thread:        2  minifb

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called
```

bunnie, 2023-04-27:
Genuinely curious: why would an OS not allow drawing to screen from threads other than thread #1? Seems like an arbitrary distinction that adds a lot of overhead to enforce. I can see restricting which processes having access, but once you have threads setup you're all in the same sandbox anyways, right? You either trust the program or you don't...

xobs, 2023-04-27:
Graphics libraries tend to be huge balls of mutable state, and aren't thread-safe. To get around this, they usually just say you're only supposed to access drawing from one thread.

I see warnings from Android telling you to not call UI functions from a child thread. WinForms/UWP also don't like it when you do that, and it can put the program into an undefined state. Qt will likewise behave badly if you try to paint widgets across threads.

Macos seems to go one step further and calls `abort()` if a UI function is called from a non-UI thread, and since the UI is started before `main()` is called the main thread is the UI thread.

Incidentally, the "huge ball of mutable state" is one of the reasons Rust has such a bad GUI story -- Rust *needs* to be thread-safe in order to compile, and most of the other approaches won't even compile.

bunnie, 2023-04-27:
ugh why does this auto-close

bunnie, 2023-04-27:
https://github.com/betrusted-io/xous-core/commit/9022fe68270194148a59cfa777bf5c6a1a30367c is up, @xobs can you try to re-test for macos

bunnie, 2023-04-27:
I wonder if the "macos" exception shouldn't be here on this line:

https://github.com/betrusted-io/xous-core/blob/9022fe68270194148a59cfa777bf5c6a1a30367c/services/graphics-server/src/main.rs#L100-L112

Not sure why it is there.

bunnie, 2023-04-27:
(@xobs can you try removing that and seeing it if fixes the compile issue for you)

bunnie, 2023-04-27:
I want you to remove it. So it should just say `#[cfg(any(not(feature = "ditherpunk")))] `

xobs, 2023-04-27:
With that change, it fails with:

```
error[E0425]: cannot find function `claim_main_thread` in module `backend`
  --> services/graphics-server/src/main.rs:99:14
   |
99 |     backend::claim_main_thread(move |main_thread_token| {
   |              ^^^^^^^^^^^^^^^^^ not found in `backend`

error[E0412]: cannot find type `MainThreadToken` in module `backend`
   --> services/graphics-server/src/main.rs:114:45
    |
114 | fn wrapped_main(main_thread_token: backend::MainThreadToken) -> ! {
    |                                             ^^^^^^^^^^^^^^^ not found in `backend`

error[E0061]: this function takes 0 arguments but 1 argument was supplied
   --> services/graphics-server/src/main.rs:119:23
    |
119 |     let mut display = XousDisplay::new(main_thread_token);
    |                       ^^^^^^^^^^^^^^^^ -----------------
    |                                        |
    |                                        unexpected argument
    |                                        help: remove the extra argument
    |
note: associated function defined here
   --> services/graphics-server/src/backend/minifb_macos.rs:41:12
    |
41  |     pub fn new() -> XousDisplay {
    |            ^^^

Some errors have detailed explanations: E0061, E0412, E0425.
For more information about an error, try `rustc --explain E0061`.
error: could not compile `graphics-server` due to 3 previous errors
warning: build failed, waiting for other jobs to finish...
Error: "Local build failed"

Core on  main [$!?] via 🐍 v3.10.9 via 🦀 v1.69.0 took 20s 
❯                                                                        (base) 
```

bunnie, 2023-04-27:
Can you confirm your mod.rs on backend looks like this:

https://github.com/betrusted-io/xous-core/blob/9022fe68270194148a59cfa777bf5c6a1a30367c/services/graphics-server/src/backend/mod.rs#L1-L4

(i.e. that you're up to date with `main`)

there was a patch previously that selected out the backend to a macos specific one

xobs, 2023-04-27:
That does fix it, yes. I didn't realise there were additional patches on `main`

It works with those patches, and the patch you suggested earlier.

bunnie, 2023-04-27:
cool if it it's fixed for you then please close the issue.

PR #374 (2023-04-27): graphics-server: Run `minifb` event loop on main thread
Fixes #373
closed 2023-04-27 by xobs