betrusted-io/xous-core #376: Device re-awakens and freezes during kernel load

plaurent, 2023-04-29:
**Setup**
- flash bleeding edge
- create image with `cargo xtask app-image vault mtxcli transientdisk ball`

**Steps to replicate**
1. boot and unlock (allow device to connect to Wifi)
2. Switch to app `vault`
3. Switch back to `shellchat`
4. Sleep now

**Result**: Device appears to sleep normally but within 2 minutes -- device should reboot and freeze on "Loading kernel..." kernel (~90 seconds).

Serial capture: [possession-log.txt](https://github.com/betrusted-io/xous-core/files/11358309/possession-log.txt)

Notes:
- it appears to be necessary to connect to Wifi.  If I delete the wifi AP connection information I cannot reproduce.
- it appears to be necessary to have more than just Vault installed.  If I just have vault I cannot reproduce.
- On my custom Xous app-image, this reboot and freeze happens within 10 seconds; happy to share a fork/branch if that would help.

[image]

bunnie, 2023-04-29:
oh that serial console log is very interesting. it's not at all what I had expected.

do you happen to note which process is PID 5? it would show up in the build log.

could you please link me to your custom build?

bunnie, 2023-04-29:
ah, on my system, pid 5 is xous-susres. so, the system is crashing during resume inside the susres manager directly. 

this is probably a memory alignment issue, or caching, or maybe something to do with some of the recent kernel changes.

definitely a link to your repo so I can reproduce exactly would be helpful, thanks!

plaurent, 2023-04-29:
@bunnie To clarify — the crash log I posted was using `xous-core` `main` (not my repo).  With that repo, reboot-then-crash happens about 90 seconds after "Sleep now".

But if you'd like to use my repo (the only difference being the reboot-then-crash seems to happen about 10 seconds after "Sleep now") — that repo is here: https://github.com/plaurent/xous-core/tree/personal/deploy/apps/book

and my app-image includes `vault transientdisk mtxcli book`

plaurent, 2023-04-29:
The table of process IDs for the app-image (the one I used when reporting the issue and recording the serial output, from `xous-core/main`) was:

```
31    process names:
        PID 1: kernel
        PID 2: xous-ticktimer
        PID 3: xous-log
        PID 4: xous-names
        PID 5: xous-susres
        PID 6: graphics-server
        PID 7: keyboard
        PID 8: spinor
        PID 9: llio
        PID 10: com
        PID 11: net
        PID 12: dns
        PID 13: gam
        PID 14: ime-frontend
        PID 15: ime-plugin-shell
        PID 16: codec
        PID 17: modals
        PID 18: root-keys
        PID 19: trng
        PID 20: sha2
        PID 21: engine-25519
        PID 22: jtag
        PID 23: status
        PID 24: shellchat
        PID 25: pddb
        PID 26: usb-device-xous
        PID 27: vault
        PID 28: mtxcli
        PID 29: transientdisk
```

bunnie, 2023-05-03:
Hm, this really does smell like an OOM issue. Instead of building with `cargo xtask app-image` as the verb, can you try `cargo xtask app-image-xip`? This moves some servers to disk, freeing up some RAM. Obviously, the apples-to-apples comparison would be needed in terms of number of apps loaded.

bunnie, 2023-05-07:
Thanks for confirming that. I think this does sound like an OOM issue then.

Basically, Xous with the default build setting loads all the servers into RAM, which eats away at the available RAM to operate. Somehow on suspend/resume there is either a peak demand for RAM (or maybe possibly a memory leak?) that can't be met and the system crashes when you have too many apps loaded.

I think the action is I am going to set `app-image-xip` as the default for the upcoming release, and I will also update the wiki docs to refer to it. This will allow for more apps to be loaded and hopefully avoid this issue.

bunnie, 2023-05-07:
Updated here: https://github.com/betrusted-io/xous-core/commit/9bc7d6b7e619fea20cfaaf5b4251198fab159c72

Couldn't find other places where it's referred to, if anyone is using a different spot for a reference please drop a comment so I can update that too.

bunnie, 2023-05-21:
I'm going to close this issue assuming that the OOM explanation is in fact the root cause, since I have not heard any further activity on this. If the problem resumes or manifests again, please re-open the issue!

closed 2023-05-21 by bunnie