betrusted-io/xous-core #381: Possible corruption of ProcessStartup data during CreateProcess syscall on RISC-V

vihaanjim, 2023-05-12:
I've been trying to recreate the test-spawn program with the following code:
```
let stub = include_bytes!("spawn-stub");
let args = xous::ProcessArgs::new(stub, xous::MemoryAddress::new(0x2050_1000).unwrap(), xous::MemoryAddress::new(0x2050_1000).unwrap());
let spawn = xous::create_process(args).expect("Couldn't create spawn process");
log::info!("Spawn PID: {}, Spawn CID: {}", spawn.pid, spawn.cid);

// perform a ping to make sure that spawn is running
let result = xous::send_message(
    spawn.cid,
    xous::Message::new_blocking_scalar(4, 1, 2, 3, 4),
)
    .unwrap();
log::info!("Result of ping: {:?}", result);
```
However, this fails with `ServerNotFound` since `spawn.cid` is a number greater than 32 (it varies, but most of the time I get a CID of 4124 and a PID of 6).
I have determined that here (found in `handle_inner`)
```
SysCall::CreateProcess(process_init) => SystemServices::with_mut(|ss| {
    ss.create_process(process_init)
          .map(xous_kernel::Result::NewProcess)
}),
```
the correct values (a CID of 6 and a PID of 29 on my system) are present, but by `create_process` in `xous-rs`, the values are incorrect.
I believe that the last point where the values are correct are here (in `trap_handler` in kernel/src/arch/risv/irq.rs)
```
ArchProcess::with_current_mut(|p| {
    let thread = p.current_thread();
    // If we're resuming a process that was previously sleeping, restore the
    // thread context. Otherwise, keep the thread context the same and pass
    // the return values in 8 argument registers.
    if response == xous_kernel::Result::ResumeProcess {
        crate::arch::syscall::resume(current_pid().get() == 1, thread);
    } else {
        // println!("Returning to address {:08x}", thread.sepc);
        unsafe { _xous_syscall_return_result(&response, thread) };
    }
});
```
I have verified that spawn is actually running by manually connecting to CID 6, so my guess is that the error must be occurring when the `ProcessStartup` is copied into the registers during `_xous_syscall_return_result`, or later when it is retrieved.

vihaanjim, 2023-07-02:
I now notice that I also sometimes get a PID of 6 and a CID of 29, which is the opposite of what I should be getting (PID of 29 and CID of 6). Perhaps this is part of the problem?
I should also mention that I've only tested this on renode so far, so perhaps the problem is specific to renode.

bunnie, 2023-07-05:
tagging @xobs as he would be best suited to answer this.

@vihaanjim do you happen to have your test case checked into a repo somewhere so it can be easily reproduced?

vihaanjim, 2023-07-06:
> do you happen to have your test case checked into a repo somewhere so it can be easily reproduced?

The issue happens with test-spawn as well (following patch should turn it into a test file). For some reason the kernel terminates the process on a panic in test mode instead of allowing the test to fail, but using the debugger the test does indeed panic at `assert!(process.cid < 32, "CID={}>=32", process.cid);` because the CID is 1937075975.
```
diff --git a/services/test-spawn/src/main.rs b/services/test-spawn/src/main.rs
index e399363f..c5e4d547 100644
--- a/services/test-spawn/src/main.rs
+++ b/services/test-spawn/src/main.rs
@@ -1,3 +1,4 @@
+#[cfg_attr(test, test)]
 fn main() {
     println!("Loading stub file");
     let spawn_stub = include_bytes!("spawn-stub");
@@ -6,8 +7,10 @@ fn main() {
     let args = xous::ProcessArgs::new(
         spawn_stub,
         xous::MemoryAddress::new(0x2050_1000).unwrap(),
+	xous::MemoryAddress::new(0x2050_1000).unwrap(),
     );
     let process = xous::create_process(args).unwrap();
+    assert!(process.cid < 32, "CID={}>=32", process.cid);
     println!(
         "Connected to process. PID: {:?}, CID: {:?}",
         process.pid, process.cid
@@ -15,8 +18,8 @@ fn main() {
     let result = xous::send_message(
         process.cid,
         xous::Message::new_blocking_scalar(4, 1, 2, 3, 4),
-    )
-    .unwrap();
+    );
+    assert!(result.is_ok(), "Could not connect to CID {}. Got: {:?}", process.cid, result);
     println!("Result of ping: {:?}", result);
     println!("Hello, world!");
 }
```

closed 2023-08-08 by vihaanjim

bunnie, 2023-08-08:
thanks for fixing this, it's really subtle!