betrusted-io/xous-core #382: bug: 0.9.13 lock device (reboot) with usb connected infinite loop

wizzard0, 2023-05-19:
Xous version: 0.9.13

**Expected behavior:** Reboot/Lock works with USB connected
**Actual:** "lock device (reboot)" flashes the loaders then ends with an infinite loop of "cant power off while charging" instead of asking for the unlock pin

After removing the cable the device sleeps, then asks for the pin after f1+f4

gsora, 2023-05-19:
Aborting the sleep procedure and going straight into unlock when USB is connected after sleep/reboot seems sensible in this case.

WDYT?

wizzard0, 2023-05-19:
Agree completely (the main use case is to lock, not to sleep). The sleep is better done after disconnect w/timeout or something in #383

gsora, 2023-05-19:
So something like this:

- user selects sleep from the main menu
- "can't sleep while charging" modal appears, which is then dismissed
- user tries again with power disconnected
- reboot goes through
    - no cable connected ever, so device boots and then sleeps
    - cable was connected after the reboot happens, at that point "can't sleep while charging" modal appears, which is then dismissed and the standard PIN login flow appears, device doesn't sleep

I would still keep the reboot->sleep flow as it is now, just add some ornament to make the UX less confusing.

bunnie, 2023-05-21:
Hm, I kind of don't like how GH just closes the issue without copying the comment here for discussion. So, I am mirroring it here:

this should fix the issue on https://github.com/betrusted-io/xous-core/issues/382. The behavior is now:

*device not plugged in*

1. select "Lock device (reboot)"
2. device reboots
3. device goes to sleep
4. user wakes up the device
5. user is challenged with the unlock PIN

*device is plugged in*

1. select "Lock device (reboot)"
2. device reboots
3. user is challenged with the unlock PIN

I think this should be the "least surprising" outcome given
the description of the menu item.

Please re-open if this is not the correct resolution.

closed 2023-05-21 by bunnie

wizzard0, 2023-05-21:
Yes this looks good! Gonna try to apply this to device and check

commit fe646f52 (2023-05-21): fix #382

this should fix the issue on #382. The behavior is now:

*device not plugged in*

1. select "Lock device (reboot)"
2. device reboots
3. device goes to sleep
4. user wakes up the device
5. user is challenged with the unlock PIN

*device is plugged in*

1. select "Lock device (reboot)"
2. device reboots
3. user is challenged with the unlock PIN

I think this should be the "least surprising" outcome given
the description of the menu item.