betrusted-io/xous-core #383: FR: sleep on timeout, or after USB disconnect

wizzard0, 2023-05-19:
Scenario: 

- Connect precursor to charge
- Disconnect (it's unlocked bc it can't sleep while charging), put in pocket, forget to explicitly enable sleep
- Pull out of pocket, discover it's discharged again

Possible options:
- Allow to lock while charging
- Sleep/lock after some timeout after USB disconnect

bunnie, 2023-05-21:
From the commit above:

    Adds a menu item (defaults to 0 or off) for autosleep timeout in
    minutes.

    Once the autosleep timeout is hit, the system tries to suspend.

    - It will not try to sleep if the system is plugged in.
    - It will reset the timer if the power state changes (plug/unplug).
    - It will reset the timer every time a key is pressed.
    - The timer is reset on suspend, so it does not try to immedaitely re-sleep on wake.

    The autosleep handler is domiciled in the "pump" loop which is
    responsible for updating the uptime timestamp in the status bar.
    It is pinged at a 1 second interval.

    Therefore, the sleep timeout has a +/-1 second variability in it,
    and also, a quick plug-unplug event (< 1 second) will likely go
    unnoticed.

bunnie, 2023-05-21:
Note that one major caveat is that the system will *not* autosleep if you reboot-to-lock, because the sleep interval is stored in the PDDB.

Without access to it, it will default to no autosleep!

commit 6f9febcc (2023-05-21): this resolves #383

Adds a menu item (defaults to 0 or off) for autosleep timeout in
minutes.

Once the autosleep timeout is hit, the system tries to suspend.

- It will not try to sleep if the system is plugged in.
- It will reset the timer if the power state changes (plug/unplug).
- It will reset the timer every time a key is pressed.
- The timer is reset on suspend, so it does not try to immedaitely re-sleep on wake.

The autosleep handler is domiciled in the "pump" loop which is
responsible for updating the uptime timestamp in the status bar.
It is pinged at a 1 second interval.

Therefore [... cut]
closed 2023-05-21 by bunnie