betrusted-io/xous-core #385: FR: Selectable virtual FIDO keys

wizzard0, 2023-05-19:
(0.9.13) Allow to select which virtual FIDO key to use

**Use cases:**
- test fido2 auth scenarios that require multiple keys for one account
- multiple accounts on the same website with different keys, where the website can't deduce they belong to the same physical Precursor device and can't discover them without the corresponding PDDBs being unlocked

**Currently**, as per the discussion from chat:

bunnie> the identity key is set up in the system basis, so it's always there no matter what secret basis is open

gsora> having your own attestation key makes your specific device stand out from the crowd
gsora> citing the spec: Every U2F device has a shared 'Attestation' key pair which is present on it - this key is shared across a large number of U2F device units made by the same vendor (this is to prevent individual identifiability of the U2F device). Every public key output by the U2F device during the registration step is signed with the attestation private key.

**Proposed UX**:
wizzard0> I see. I want a per-credential `device private key` then. Something like "`select a virtual token to use`" if it has to be chosen before the protocol starts and the website doesn't send any tokens/hashes first which could be used to detect which cred to use (which cred they are related to)

spoelstraethan, 2023-09-08:
@wizzard0 I think that selection UI may violate the PDDB "zero knowledge across bases" aspect of the Precursor. If you are worried about a specific identity being seen/selected you need to unmount that basis and mount the correct one before accessing the site.

wizzard0, 2023-09-08:
@spoelstraethan No, I don't think so

1) right now there's no way to dismount the fido identity basis, as it always uses the System one, IIUC

2) same as you pick a password to autotype, instead of trying to bruteforce them all - you would pick an identity to authenticate

spoelstraethan, 2023-09-08:
Hmm, it may depend if you are talking U2F or FIDO or FIDO2, because when I don't have a secret basis (or bases) mounted their entries don't show up on the FIDO page. I haven't tested whether I can authenticate with an identity that lives in a basis without mounting it but I should try that.