betrusted-io/xous-core #386: 0.9.13 F1-F4 keys insert garbage in the hosted emulator

wizzard0, 2023-05-19:
As of 3aeb360, pressing F1-F1 and Fn+LeftArrow in hosted mode also adds some random characters:

os: macOS Ventura 13.3.1
rust: rust 1.69.0
xous-core: 0.9.13 3aeb360

F1-F4
```
INFO:keyboard: injecting key ''(11) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(f704) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(12) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(f705) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(13) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(f706) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(14) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(f707) (services/keyboard/src/main.rs:941)
INFO:com: HOSTED: set backlight to 0,0 (services/com/src/main.rs:676)
```
Fn+LeftArrow, Down, Up, Up, Fn+LeftArrow (Close menu):
```
INFO:keyboard: injecting key '∴'(2234) (services/keyboard/src/main.rs:941)
INFO:gam::contexts: allow_mainmenu: true (services/gam/src/contexts.rs:612)
INFO:keyboard: injecting key ''(f729) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key '↓'(2193) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(f701) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key '↑'(2191) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(f700) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key '↑'(2191) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(f700) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key '∴'(2234) (services/keyboard/src/main.rs:941)
INFO:keyboard: injecting key ''(f729) (services/keyboard/src/main.rs:941)
```

bunnie, 2023-05-21:
Interesting. Did a little searching on the issue and it looks like this is a common problem for OSX clients:

https://bugs.mojang.com/browse/MC-49755

Seems OSX injects some extra private use characters after sending over the key codes. According to Wikipedia:

> [Apple](https://en.wikipedia.org/wiki/Apple_Inc.) lists a range of 1,280 characters in its developer documentation[[14]](https://en.wikipedia.org/wiki/Private_Use_Areas#cite_note-15) from U+F400–U+F8FF within the PUA for Apple's use. Of those, only 311 are used, in the range U+F700–U+F8FF ([NeXT](https://en.wikipedia.org/wiki/NeXT) ([NeXTSTEP](https://en.wikipedia.org/wiki/NeXTSTEP) and [OPENSTEP](https://en.wikipedia.org/wiki/OPENSTEP)) and [Apple](https://en.wikipedia.org/wiki/Apple_Inc.) (Mac OS X AppKit)).[[15]](https://en.wikipedia.org/wiki/Private_Use_Areas#cite_note-PUA_CORPCHAR-16)

The PR looks correct code-wise in terms of restricting, but can we bracket the ignored area to just the private use area specified by Apple? So instead of throwing away everything with 0xFxxx, I think we should just reject the 0xF700-0xF8FF range. This allows us to interoperate with other perhaps correctly entered unicode characters in the 0xFxxx range.

bunnie, 2023-05-21:
Copied comment to PR

wizzard0, 2023-05-21:
Agree, narrowed down to the 0xf700-0xf8ff, rebased in https://github.com/betrusted-io/xous-core/commit/e08f4b89c55531179f9a12546d36b459ed13ec75

PR #387 (2023-05-21): Fix #386: Ignore key codes over 0xf000 on insert
This closes #386 by ignoring any keys over 0xf000

Not sure this is a completely correct logic. Checked with key codes that seem to be injected on Latin, Cyrillic and japanese Kana keyboard on macOS Ventura.
closed 2023-05-21 by bunnie