betrusted-io/xous-core #407: "Ethernet address dd-dd-dd-dd-dd is not unicast" Rust panic

RaitoBezarius, 2023-08-28:
Hello there,

I was doing a SoC upgrade, finished, then my system started an EC autoupdate and in the meantime I was doing the full SoC reset.

At restart time, I got this "Guru Meditation" window showing me with a panic trace:

> "Ethernet address dd-dd-dd-dd-dd is not unicast"

From smoltcp.rs (interfaces.rs:1118)

I believe this may be related to the "reset during EC autoupdate", I am not sure: is there a way to recover from that?

I did not attach the debug hat yet but I am familiar with all of that, just curious if I can recover easily without it.

bunnie, 2023-10-03:
#419 includes a patch that checks the MAC address passed in from the EC driver layer and generates a random MAC in the case that it is not correct. 

I'm not quite sure what your question is, though: in terms of recovering from the error, a reboot might be enough to work around the situation (usually this is caused by the EC being in reset or in a bad state). But, if the EC is actually mis-programmed or did not update correctly, you would have to use the fail-safe method (with a JTAG cable) to recover the EC. 

I'll close this issue now since the root cause has been addressed, but curious if you were able to overcome your problem.

closed 2023-10-03 by bunnie

RaitoBezarius, 2023-10-03:
Didn't had time to perform the fail-safe JTAG recovery, will do so at some point, thanks for the tip!

RaitoBezarius, 2023-10-14:
> #419 includes a patch that checks the MAC address passed in from the EC driver layer and generates a random MAC in the case that it is not correct.
>
> I'm not quite sure what your question is, though: [...]

So TL;DR: tried failsafe, encountered https://github.com/betrusted-io/xous-core/issues/113, tried to use config_up5k.sh, restored the EC, rebooted into the OS without this issue, then I did a "force gateware update", bricked it again, provision_xous worked (!!!!), reprovisioned until the **kernel**, at kernel time, it failed utterly with the 113 issue again.

Overall, I am satisfied :D.