betrusted-io/xous-core #424: usb strangeness with openbsd host

eau-u4f, 2023-10-08:
trying to plug the precursor on an openbsd machine through USB, I get the following on the host machine kernel log:
uhub1: device problem, disabling port 5

not sure what this means since I did not really investigate deeper right as traveling, does that ring a bell?

openbsd line in kernel:
https://github.com/openbsd/src/blob/master/sys/dev/usb/uhub.c#L604

if not, I guess, I'll dig and/or recompile kernel with debug to track down what's happening.. in the following days..

qbit, 2023-10-09:
Have you tried another cable? Is this in just regular mode on the precursor (not trying to do auto-type or anything?)?

qbit, 2023-10-09:
On OpenBSD current I get the following when I connect the cable:
```
ugen0 at uhub0 port 2 "Foosn Precursor pvt2" rev 2.00/1.01 addr 2
```

And in "vault mode" I get:
```
uhidev0 at uhub0 port 2 configuration 1 interface 0 "Kosagi Precursor" rev 2.10/0.10 addr 2
uhidev0: iclass 3/0
fido0 at uhidev0: input=64, output=64, feature=0
uhidev1 at uhub0 port 2 configuration 1 interface 1 "Kosagi Precursor" rev 2.10/0.10 addr 2
uhidev1: iclass 3/1
ukbd0 at uhidev1: 128 variable keys, 0 key codes
```

eau-u4f, 2023-10-09:
which release of OpenBSD, which release of xous? I am running the latest stable xous and OpenBSD 7.4
I triggered usb debug to be able to use the precursor updater

qbit, 2023-10-09:
OpenBSD 7.4 hasn’t been released yet. As I said, I am on current. Latest FWs on the precursor.

if your showing 7.4 you are running snapshots (current).

Did you try a new cable? Perhaps your usb port isn’t supplying enough voltage?

> I triggered usb debug

I don’t know what you mean by this.

qbit, 2023-10-09:
`usbdevs -vv` when running `Shellchat`:
```
addr 02: 1209:5bf0 Foosn, Precursor pvt2
         full speed, power 100 mA, config 1, rev 1.01
         driver: ugen0
```

qbit, 2023-10-09:
in `Vault`:
```
addr 02: 1209:3613 Kosagi, Precursor
         full speed, power 100 mA, config 1, rev 0.10, iSerial 1a452a193ea85c
         driver: uhidev0
         driver: uhidev1
```

eau-u4f, 2025-09-14:
ok, I've retried multiple times, the devices appears only when I "reboot" (Lock Device) the device WHILE it is plugged, if I unplug and replug the device it does not come back, it appears like this:

addr 02: 1209:5bf0 Foosn, Precursor pvt2
	 full speed, power 100 mA, config 1, rev 1.01
	 driver: ugen2

no iSerial like you, I am running -current  ( soon to become 7.8) 

However switching to vault, it appears as:

addr 02: 1209:3613 Kosagi, Precursor
	 full speed, power 100 mA, config 1, rev 0.10, iSerial 7b81............
	 driver: uhidev0
	 driver: uhidev1

bunnie, 2025-09-14:
I'm not quite following so I will say what i think you are reporting and you can tell me if I am reading the issue correctly.

What you are reporting is if you plug the device in while it is powered on and showing the vault app, it does not enumerate at all on BSD. However if you have it plugged in *and* you put it through a lock cycle, it can enumerate?

A piece of state that i am trying to tease out from your scenario is that the Precursor's descriptors change between the default, on reboot descriptors, and the first time you switch into 'vault'. Between those two state changes you effectively show the host a different USB device (although the vid/pid stays the same - so if the bsd kernel is caching usb configs based on vid/pid that could be a problem), so I'm trying to figure out which of the two states is causing trouble.

eau-u4f, 2025-09-14:
Ok sorry for my bad reporting, I will try to be better.

Yes, when I plug the device, it does not show up as a USB devices, when I go through the Lock cycle, it shows up as USB devices.

When I switch to vault, it shows up as new USB device.
When I switch back to shellchat no usb device.

Now:
I go through a USB cycle to attempt a backup, it shows up, as I prepare the device for backup, the USB device disappears.

bunnie, 2025-09-14:
ah ok. this sounds like an issue then with the debug core. so inside precursor (details are fuzzy its been a while) there are actually two USB engines. one is called the debug core and it requires no device code to operate - it is supposed to work even if there is no code in the device. It's what is available by default.

There is a second Spinalhdl core. This requires an OS but it can emulate all kinds of devices.

The one that is causing trouble is the debug core.

Does BSD report anything at all? an error message or any clue like that? perhaps a complaint about a malformed descriptor? do you have access to a USB analyzer?

One thing about the debug core is that it presents no endpoints... it does all its work using configuration packets, iirc. It's a bit of a hack but that is how it can work without drivers or OS. If BSD is particularly strict about USB security it might be blocking that channel because it is not illegal by the spec but highly unusual.

Another possibility is iirc I provided a way to disable the debug core with a setting. I don't remember exactly when that plays in but i think i made it pretty aggressive, as it was intended for people who wanted to thwart evil maids at the expense of convenience (you can't debug, do backups etc). Do you remember configuring anything that might have turned that off? I'm not at my desk at the moment so i can't look at the code, so I'm just going by memory here...

eau-u4f, 2025-09-14:
ok let's see if this is useful for the information you would want to see with a USB analyzer, OpenBSD provides the ability to sniff USB devices for debug purposes using just tcpdump.

attached is a pcap dump, readable with wireshark of the precursor off and being plugged in on the usb port.

let me know if this is data we can use to debug the issue, before trying, otherwise I think I have a USB analyzer somewhere at home, on my pile of "not started" TODOs (learn USB).

here is the pcap capture:

[usb_boot_precursor.pcap.gz](https://github.com/user-attachments/files/22320832/usb_boot_precursor.pcap.gz)

eau-u4f, 2025-09-14:
"turned that off" , turned what off? 
some settings that would force BSD to be strict about USB ?
I did not change any debug core settings as far as I remember, but is it a setting in precursor? I do not see anything related to debug core in the preferences.

I might have misunderstood your question.

About BSD it does report the following when I plug the precursor, depending on the moment I plug it in:
uhub3: device problem, disabling port 1
uhub3: device problem, disabling port 1

For example:

TEST 1: power on
Initial State: shutdown
Kernel message:
--
ugen2 at uhub3 port 1 "Foosn Precursor pvt2" rev 2.00/1.01 addr 2
--
Reported device:
addr 02: 1209:5bf0 Foosn, Precursor pvt2
	 full speed, power 100 mA, config 1, rev 1.01
	 driver: ugen2

TEST 2: unplug precursor and replug it to the same port
Initial State: powered on (from TEST1)
Kernel message:
--
ugen2 detached
uhub3: device problem, disabling port 1
--
Reported device:
none.