betrusted-io/xous-core #444: Add `xtask` flag to enable gdb-stub

xobs, 2023-10-30:
When building the kernel, the `gdb-stub` feature can be enabled to support debugging processes both live and after they have crashed. Right now this feature needs to be manually enabled (e.g. built with something like `cargo build --release --target riscv32imac-unknown-xous-elf --package xous-kernel --features renode --features gdb-stub`), but it would be nice to add an `xtask` flag to enable this as part of the normal build.

bunnie, 2023-10-30:
This currently does work:

`cargo xtask app-image-xip --gdb-stub mtxchat --feature efuse --feature tls`

It produces these command lines:

`Command: cargo build --release --target riscv32imac-unknown-xous-elf --package xous-ticktimer --package xous-log --package xous-names --package xous-susres --package graphics-server --package early_settings --package keyboard --package spinor --package llio --package com --package net --package dns --package gam --package ime-frontend --package ime-plugin-shell --package codec  --package modals --package root-keys --package trng --package sha2 --package engine-25519 --package jtag --package status --package shellchat --package pddb --package usb-device-xous --features timestamp --features efuse --features tls --features mass-storage --features precursor --features utralib/precursor-pvt`

and 

`cargo build --release --target riscv32imac-unknown-xous-elf --package xous-kernel --features gdb-stub --features precursor --features utralib/precursor-pvt`

and

`cargo build --release --target riscv32imac-unknown-xous-elf --package loader --no-default-features --features precursor --features utralib/precursor-pvt`

In other words, `--features gdb-stub` is passed only into the kernel.

Since this feature is already in `xtask`, is this issue more about fixing the help message since it's missing there? Or is there a problem with the implementation of that itself.

bunnie, 2023-10-30:
Actually it is also printed in the help message:

```
The [cratespecs] list is treated as apps or services based on the context of [verb]. Additional crates can
be merged in with explicit app/service treatment with the following flags:
 [--app] [cratespec]     [cratespec] is treated as an additional app
 [--service] [cratespec] [cratespec] is treated as an additional service

[--lkey] and [--kkey]    Paths to alternate private key files for loader and kernel key signing (defaults to developer key)
[--no-timestamp]         Do not include a timestamp in the build. By default, `ticktimer` is rebuilt on every run to encode a timestamp.
[--no-verify]            Do not verify that local sources match crates.io downloaded sources
[--gdb-stub]             Build the kernel with GDB support
```

xobs, 2023-10-30:
My mistake. I've updated the book with that information.

closed 2023-10-30 by xobs