betrusted-io/xous-core #491: Refactor engine-sha512

kotval, 2024-01-29:
In work on upgrading curve25519-dalek, we need to update the sha2 crate to version `0.10` (ideally the most recent one `0.10.8`. We vendored that package into engine-sha512 starting at a relatively old commit. It would make this upgrade and future maintenance work much easier to instead fork https://github.com/RustCrypto/hashes, and put our hardware acceleration code there as then syncing from upstream would be facilitated better by git. If we are careful with cfg flags, we can make hardware acceleration and hosted mode both work with minimal changes to upstream.

bunnie, 2024-01-29:
Alright, I at least have what I think is a system where the dependencies are set up so I can see what has to change to go to 0.10.8 from the current 0.9.8 based on the all the errors that rustc throws: Look do-able, but it will take some doin'.

The biggest flag I see is that the ed25519-dalek crate that's being used in rootkeys is at version 1.0.1 that pulls in a sha2-0.9.9, which means that ... I *think* we lose some performance during root keys setup and re-signing operations until the ed25519-dalek crate can also be upgraded. However, upgrading this looks like a true nightmare because you enter `zeroize` conflict hell, and I'd rather defer that to later. This is strongly mitigated by the fact that we're using a pre-hash to sign the kernel, so, the "big hash" is actually not even computed -- maybe it's not even a performance issue? we'll find out.

It does look like on the move from 0.9.8 to 0.10.8, there were some substantial API changes that will roll over into regressions that hit a large number of crates inside the Xous tree. I think I'll have to split this into two phases, one is just familiarizing with the new 0.10.* series APIs and figuring out how to wedge the hardware accelerator in there, and then, figuring out how to un-break all the code that uses the 0.9.* API to work with the newer version. And then, of course, testing to make sure I haven't broken anything.

There's also an interesting quirk that we now have an out-of-tree service, which refers to versions of Xous, Xous-Api-Names, etc., which won't get automatic version-pin upgrades. I'm mentioning this here because there might need to be a follow-on issue to upgrade the helper script that maintains the version pins to also include the now out-of-tree hash crate. I think it's not too hard to extend the current maintenance script to also check the directory that contains the hasher, it just needs to be remembered to be done.

I do think this is a substantial task. I would set an expectation of 1-2 weeks, it's not going to happen overnight, but the good news is I think the overall strategy suggested by @kotval to fork the hashes crate and then patch our implementation into it should work. Although, I will note that on an API change as substantial as 0.9 to 0.10 -- basically a full rewrite of the internals -- it'll be a lot of work. That being said it looks like maybe the rate of changes isn't so high on the sha2 crate and just being able to pull in the non-breaking revs with ease is worth it.

And, of course, being at a modern version of the API so we're compatible with e.g. Signal.

bunnie, 2024-01-30:
Adding a note from chat: digest needs to go to a 0.10, rand_core to a 0.6.4, and sha2 to a 0.10.

Have to pull in rand_core for the refactor, will try to target 0.6.4.

bunnie, 2024-01-30:
Making some notes as I do the port...

The 0.10.* API is a regression for us. I can see why the implementers ... thought it was a good idea. They got clever and found a way to  use traits to re-implement a lot of the boilerplate associated with variants of a hash.

The problem is, by wrapping the structure in so many traits, it's also not possible to extend the functionality without forking the trait library and causing other chaos. In particular, a feature I rather liked about our previous implementation is that we could stipulate on a hash whether it should use software, hardware-only, or try to use hardware and then fallback to software if it's not available. Functionally they are all the same, but it turns out that a software-only hasher is going to be more performant for very tiny amounts of data (dozens of bytes); and a hardwvare-only hasher can be worth waiting several seconds to finish up a prior operation if you're hashing megabytes of data.

The new trait-based hash deriver does make things a bit cleaner; there are fewer macros in the implementation, and it feels nicer to code with, but because of the way traits work in Rust, I'm no longer able to extend the hash object with a feature to control its modes without also modifying the trait crate.

I think this means ... I'm just going to have to stick with one mode of operation, which is to try hardware then fallback to software, which means that some things will get slower. It's probably not a *huge* deal -- the biggest impact will be on the key commitments and password hashing routines where you're usually just hashing tiny amounts of data. Fortunately I don't see this in any inner loops, but still, sort of annoying that they removed an opportunity for seamless extensibility of the class with extra knobs without having to fork more crates...

Might be possible to work around it with some sort of ... shared static mutable unsafe flag to change modes, but I think that's too yucky to be worth it.

bunnie, 2024-01-30:
Note: adding the message
`// FIXME: should be "wait for hardware" strategy but this does not exist in 0.10 API`
in spots where we're unable to force hardware hasher selection, in case I can find a way to work around this.

I think there is a way to do it, but it's terribly inelegant...basically involves duplicating the entire API and changing one small line to work around the trait restirctions.

bunnie, 2024-01-30:
Alright, the PR is at a point where things compile, but with plenty of warnings. But at least the system boots with the new hash API.

The actual implementation is broken, so more debugging to come. But we're at least compiling and booting again.

Note that boot time is substantially slower for now because ed25519 no longer uses the hardware accelerated hash to check the gateware signature. Presumably once we upgrade ed25519 to the latest version, we'll get that performance back. This will just have to be an uncomfortable awkward zone while APIs get bumped...

kotval, 2024-01-30:
I'll look at supporting both modes and see if I can come up with something which isn't icky.

bunnie, 2024-01-31:
I slept on it last night, and I think I found an even better way to deal with the API that basically involves creating a base type wrapper...I was trying to target the top level wrapper, but maybe all I need is a wrapper around my core. I'll give it a try later today, probably. It does mean you can't dynamically change the strategy of the hash but I haven't found a single case where I would want to do that yet.

bunnie, 2024-01-31:
Hmm...yah, sorry about the lack of reproducibility. It's barely buildable on my system right now. I'll get it cleaned up once the actual hash is working, I probably have a bug in converting the datatypes from the 0.10 API (they do an array of blocks, instead of an array of `u8`...which works for software algorithms that work on blocks but we have to flatten that into a page buffer to transmit to the hardware hash engine, so I have to go to a hand-written ser/de step that probably has a bug). 

Anyways. Re ugprading the other ciphers, I think that's going to be inevitable...one of the things that kept me from doing this project in the first place is once you pull one cipher up, you have to pull the rest up because they all pin to each other's dependencies, and if you use more than one cipher in a crate everyone has to upgrade in that crate. But since we're now started on the path, might as well finish walking it!

bunnie, 2024-01-31:
Alright, the PR is now working and tested. I'll go ahead and give it a merge once I've got the basic format checks clearing, but it should be compiling warning-free etc.

Probably, I can see breakage on other targets, but that's what CI will uncover for us, hopefully.

bunnie, 2024-01-31:
So, the usage of the crate is a little bit tricky. When pulling in the crate, you should specify this:
`sha2 = { version = "0.10.8" }`
however, to get hardware acceleration, you need to add these features:
```
precursor = ["sha2/precursor"]
hosted = ["sha2/hosted"]
renode = ["sha2/renode"]
```
without this feature set, you'll only get a vanilla sha2 with no acceleration.

I'm not sure how this impacts e.g. stuff running inside the `libsignal` crates -- iirc it's not possible to toggle features on crates within crates, so it may mean that all libsignal stuff runs un-accelerated without a patch or fork?

bunnie, 2024-01-31:
Alright, it looks like it's trundling along in CI pretty well now. I'm going to leave this issue open for a bit so @kotval can weigh in and let me know if this satisfies the requirements!

bunnie, 2024-02-03:
@kotval if the sha refactor seems good to you, lmk if there's another crypto API you'd like me to bring up to modern standards next. There's a bunch that need work still, but I'd like to prioritize the ones you need the most help with first. If you're stuck on the curve25519 stuff, I'm also happy to take that over.

kotval, 2024-02-04:
This looks like it satisfies the need. I am going to spend some time today to see if I can get it to build with the fork of curve25519-dalek and I'll tell you how it goes.

kotval, 2024-02-04:
Okay. Do do need some help getting `cargo xtask run --feature benchmarks` to compile, but that is orthogonal to the engine-sha512 so I am going to open another card for that one. I think we can close this issue.

kotval, 2024-02-09:
Since https://github.com/betrusted-io/xous-core/pull/493 is merged, closing this.

closed 2024-02-09 by kotval