betrusted-io/xous-core #499: Upgrade curve25519-dalek to version 4.1.1

kotval, 2024-02-04:
To support recent versions of libsignal, version 4 release of curve25519-dalek is required.
Work in progress here: https://github.com/kotval/curve25519-dalek/tree/update_to_v4

kotval, 2024-02-04:
On xous-core, the work to get this integrated is here: https://github.com/kotval/xous-core/tree/dalek_to_v4. I was able to update the use of curve25519-dalek in the wycheproof tests, but I got stuck on a cryptic issue with Zeroize that I don't fully have the context for. 

I cannot build service/root-keys due to Zeroize. It seems that Zeroize does not have the feature `derive` that it needs, but I don't seem to be able to add it, either by specifying that feature in the Cargo.toml or by manually adding `zeroize_derive` to the Cargo.toml and adjusting the code accordingly.

kotval, 2024-02-08:
Okay. Since This now builds I can start integrating libsignal. Yay. Though, when trying to enable the hardware acceleration I can't get the crate to build. Since I switched to cfg flags rather than features, it seems you can't build a crate with the placeholder engine-25519 like the our old curve25519-dalek for did. If @bunnie would like to take on the cargomancy to get this to build, I can work on getting libsignal working in sigchat with the software backend for now. I updated my main branch in my fork of xous-core so that the deps are correct, and my curve25519-dalek branch on update_to_v4 now has a build script such that if you run `cargo build --release --target riscv32imac-unknown-xous-elf` it will automatically pick the accelerated backend. To select a different one for any target you can use the `--cfg=curve25519_dalek_backend="x"` where x is any other supported backend. The nice thing about this build script is it automagically picks the correct options for hosted. We may want to consider how we identify wanting to automatically use the acceleration, but for now it is https://github.com/kotval/curve25519-dalek/blob/486cd1315f203bccb9cec350b5e76218b8ad2184/curve25519-dalek/build.rs#L89.

kotval, 2024-02-09:
Continuing commenting on the work to get hardware acceleration working in https://github.com/betrusted-io/curve25519-dalek/pull/2.

bunnie, 2024-03-24:
OK, the hardware acceleration should be working and the PR merged. I'll go ahead and close out this issue because I think we're there. Thanks for all the hard work getting this together @kotval 

I will also open a new issue to track some remaining items to be covered in terms of the locking and fall-back behaviors.

closed 2024-03-24 by bunnie