betrusted-io/xous-core #507: `ring` upgrade to 0.17

bunnie, 2024-02-11:
`ring` is now at 0.17, which means we could absorb this if it benefited other efforts in Xous.

---

## TL;DR
To facilitate any testing, you may fetch the following branches:

- https://github.com/betrusted-io/xous-core/tree/ring-0.17
- https://github.com/betrusted-io/ring-xous/tree/0.17.7-merge

You will need to make sure that the top level `Cargo.toml` patch for `ring` points to wherever you cloned `ring-xous`, and then you can build a test image with:

`cargo xtask app-image --feature simple-tls`

@nworbnhoj : `lib/tls` breaks with the latest `rustls` API at `0.22.2`. Could you please have a look at the API differences and let me know what the estimated effort level would be to upgrade the library?

@kotval / @nworbnhoj : moving forward on this depends heavily on if anyone finds their efforts on building Chat clients are held up on our version of `ring` being pinned at `0.16`. If either of you say "this would greatly improve the porting process and reduce the effort to make it happen and/or make the chat clients easier to maintain", then, that would be a significant argument to move this forward sooner than later.

However, absent any actual *demand* to move the `ring` API forward, and given that such a forward movement would break some existing libraries, I don't see a compelling reason to finish the forward-port?
 
---

## Details

I took a pass at bringing `ring-xous` to 0.17.7, which can be found here:

https://github.com/betrusted-io/ring-xous/tree/0.17.7-merge

The port can do TLS, but there are some problems that prevent this from being merged and moving this forward really depends upon how much we need the latest `rustls`. In particular:

- The `tls` library that @nworbnhoj implemented breaks when you upgrade to the latest `rustls`. I don't know at what point things broke, but it looks like they did a major refactor in how they handle locally sourced certificates. It's not a "shallow" refactor of API names shuffling around -- looks like they took a good solid whack at that process. This means the stuff in the `danger` module needs a deep cut. I don't know how it all works, so I'd appreciate it if @nworbnhoj could have a look and render an opinion on the effort level.
- There are warnings when compiling the latest ring-xous. The warnings look to actually be structural problems in `ring`:

First, they are using a deprecated method for doing prefixed exports:
```
warning: use of deprecated macro `prefixed_export`: `#[export_name]` creates problems and we will stop doing it.
   --> F:\code\ring-xous\src\arithmetic\montgomery.rs:136:1
    |
136 | prefixed_export! {
    | ^^^^^^^^^^^^^^^
    |
    = note: `#[warn(deprecated)]` on by default

warning: unused attribute `allow`
   --> F:\code\ring-xous\src\arithmetic\montgomery.rs:135:1
    |
135 | #[allow(deprecated)]
    | ^^^^^^^^^^^^^^^^^^^^
    |
note: the built-in attribute `allow` will be ignored, since it's applied to the macro invocation `prefixed_export`
   --> F:\code\ring-xous\src\arithmetic\montgomery.rs:136:1
    |
136 | prefixed_export! {
    | ^^^^^^^^^^^^^^^
    = note: `#[warn(unused_attributes)]` on by default
```
The maintainer has a `TODO` to stop doing this, but it is still pending even with the latest code:

https://github.com/briansmith/ring/blob/6762d656742d7ea431948356760da28e09292b72/src/arithmetic/montgomery.rs#L134-L136

Second, one function declaration is inconsistent:

```
warning: `CRYPTO_memcmp` redeclared with a different signature
   --> F:\code\ring-xous\src\c2rust\curve25519.rs:6:5
    |
6   | /     fn CRYPTO_memcmp(
7   | |         a: *const core::ffi::c_void,
8   | |         b: *const core::ffi::c_void,
9   | |         len: size_t,
[... 59 lines cut ...]
DBG :rustls::client::tls12: Server DNS name is DnsName("bunniefoo.com") (C:\Users\bunnie\.cargo\registry\src\index.crates.io-6f17d22bba15001f\rustls-0.22.2\src\client\tls12.rs:687)
INFO:shellchat::cmds::net_cmd: readout cipher suite (services\shellchat\src\cmds\net_cmd.rs:596)
INFO:shellchat::cmds::net_cmd: Current ciphersuite: TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (services\shellchat\src\cmds\net_cmd.rs:598)
INFO:shellchat::cmds::net_cmd: read TLS response (services\shellchat\src\cmds\net_cmd.rs:600)
INFO:shellchat::cmds::net_cmd: len: 291 (services\shellchat\src\cmds\net_cmd.rs:602)
INFO:shellchat::cmds::net_cmd: HTTP/1.1 200 OK
Date: Sun, 11 Feb 2024 09:14:29 GMT
Server: Apache/2.4.10 (Debian)
Last-Modified: Mon, 05 Jun 2017 16:46:12 GMT
ETag: "2d-551393da14d9b"
Accept-Ranges: bytes
Content-Length: 45
Connection: close
Content-Type: text/html

<html>
        Ce n'est pas une page vide.
                                   </html>
                                           (services\shellchat\src\cmds\net_cmd.rs:603)
```

I would hesitate to say "it works" at this point, it's more like "nothing is obviously broken and preventing forward progress". However, I don't want to merge this because I'm not sure if it's worth it right now. To move ahead, I think we have to satisfy two prerequisites:

- [ ] `lib/tls` has to be ported to the `rustls 0.22.2` API level
- [ ] we actually need a reason to do this work, i.e., some part of the `chat client` effort is blocked on `ring 0.17`

nworbnhoj, 2024-02-14:
Overall, the changes to `rustls` appear to be sound, and while there is a little work to upgrade `xous/libc/tls`, it seems manageable and sensible.

Some of the changes to `rustls` are a relatively straight-forward relocation of structs (`ServerCertVerified` & `WebPkiVerifier`) . The more impactful change is the abolition of `rustls::Certificate` and `rustls::OwnedTrustAnchor` in favor of `rustls::pki_types::CertificateDer`.

`rustls` v.21.7 utilised `Certificate` and `OwnedTrustAnchor`  to hold **only** the critical `X.509` fields. This was an attractive aspect for `xous` from a size/storage perspective. `rustls` v.22.2 uses a `rustls::pki_types::CertificateDer` in its operations, retaining the full `X.509` structure.

The obvious (and favored) path forward is to similarly adopt `rustls::pki_types::CertificateDer` within `xous/libc/tls` and save this structure to the `pddb`. This is obviously a (relatively minor) breaking change - and will be (somewhat) less space efficient - but the end result should be more simple code.

The obvious alternative would be to define `OwnedTrustAnchor` in `xous/libc/tls` and retain the size advantages - but this also means introducing the various translations between  `rustls::pki_types::CertificateDer` and  `OwnedTrustAnchor` that `rustls` has chosen to avoid. This is likely an unwise choice.

I will progress along these lines and submit a PR - all going well.

bunnie, 2024-02-15:
What's the size difference of holding a whole X.509 certificate versus just the critical fields? My intuition says it's on the order of 10's of kiB, so it shouldn't be a problem to retain the whole certificate. We just have to retain a handful of them too, correct? I did notice when I tried to compile against the static list of certificates from webpki crate that it add +2MiB to the binary size but I assumed we wouldn't carry that around and instead we'd just cherry pick a few certificates from that list.

Basically if the impact is on the order of 10kiB (or even low 100's kiB) I'm not too worried about the size impact. But if we are talking about +2MiB of static data being linked in I should take a closer look.

bunnie, 2024-03-01:
Looks like the `tls` feature compiles, I'll go ahead and merge this now so we're at least at ring 0.17. 

I'm going to change the title of the issue to reflect that the upgrade is done, but, we need to check in periodically on the ring development and see if the deprecation notices and warnings have been cleaned up from the code base...

bunnie, 2024-03-01:
The new TODO:

- [ ] clean up deprecation warnings, once they are cleaned up from the maintainer's crate
- [ ] clean up inconsistent method declaration, once it's fixed in the maintainer's crate

I think unless something is urgently broken, this is a low-priority task, and we'll keep an eye on `ring` 0.17 and wait until all the warnings are cleaned up. I have a feeling that something awful might be in store for the current practice used of patching all the function calls, so fixing these remaining warnings could be quite a task.