betrusted-io/xous-core #538: cargo xtask run results in white screen

rowr111, 2024-05-09:
When I try to start the emulator using `cargo xtask run` all I see is a white screen.  Example log and screenshot below.
I got latest from upstream/main and I did `rustup update` before this, FYI.

[image]

```
PS D:\code\xous-core> cargo xtask run hello
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.43s
     Running `target\debug\xtask.exe run hello`
No change to the target locale language of en
    Command: cargo build --release --package xous-ticktimer --package xous-log --package xous-names --package xous-susres --package graphics-server --package early_settings --package keyboard --package spinor --package llio --package com --package net --package dns --package gam --package ime-frontend --package ime-plugin-shell --package codec --package modals --package root-keys --package trng --package sha2@0.10.8 --package jtag --package status --package shellchat --package pddb --package usb-device-xous --package hello --features timestamp --features pddbtest --features ditherpunk --features tracking-alloc --features tls --features hosted
warning: locales@0.1.0: Reading D:\code\xous-core/**/i18n.json
warning: locales@0.1.0: Reading external D:\code\sigchat/**/i18n.json
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\app-loader\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\ball\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\hello\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\hidv2\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\mtxchat\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\mtxcli\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\repl\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\apps\\vault\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\libs\\chat\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\libs\\tls\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\services\\gam\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\services\\ime-frontend\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\services\\modals\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\services\\net\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\services\\pddb\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\services\\root-keys\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\services\\shellchat\\locales\\i18n.json"
warning: locales@0.1.0: "D:\\code\\xous-core\\services\\status\\locales\\i18n.json"
   Compiling gam v0.1.0 (D:\code\xous-core\services\gam)
   Compiling xous-ticktimer v0.1.32 (D:\code\xous-core\services\xous-ticktimer)
   Compiling modals v0.1.0 (D:\code\xous-core\services\modals)
   Compiling ime-frontend v0.1.0 (D:\code\xous-core\services\ime-frontend)
   Compiling usb-device-xous v0.1.0 (D:\code\xous-core\services\usb-device-xous)
   Compiling hello v0.1.0 (D:\code\xous-core\apps\hello)
   Compiling root-keys v0.1.0 (D:\code\xous-core\services\root-keys)
   Compiling pddb v0.1.0 (D:\code\xous-core\services\pddb)
   Compiling net v0.1.0 (D:\code\xous-core\services\net)
   Compiling userprefs v0.1.0 (D:\code\xous-core\libs\userprefs)
   Compiling dns v0.1.0 (D:\code\xous-core\services\dns)
   Compiling tls v0.1.0 (D:\code\xous-core\libs\tls)
   Compiling status v0.1.0 (D:\code\xous-core\services\status)
   Compiling shellchat v0.1.0 (D:\code\xous-core\services\shellchat)
warning: struct `StdoutTracker` is never constructed
  --> services\shellchat\src\main.rs:68:8
   |
68 | struct StdoutTracker {
   |        ^^^^^^^^^^^^^
   |
   = note: `#[warn(dead_code)]` on by default
warning: `shellchat` (bin "shellchat") generated 1 warning
    Finished `release` profile [optimized] target(s) in 31.49s
Starting hosted mode...
    Command: cargo run --release -- D:\code\xous-core/target/release/xous-ticktimer.exe D:\code\xous-core/target/release/xous-log.exe D:\code\xous-core/target/release/xous-names.exe D:\code\xous-core/target/release/xous-susres.exe D:\code\xous-core/target/release/graphics-server.exe D:\code\xous-core/target/release/early_settings.exe D:\code\xous-core/target/release/keyboard.exe D:\code\xous-core/target/release/spinor.exe D:\code\xous-core/target/release/llio.exe D:\code\xous-core/target/release/com.exe D:\code\xous-core/target/release/net.exe D:\code\xous-core/target/release/dns.exe D:\code\xous-core/target/release/gam.exe D:\code\xous-core/target/release/ime-frontend.exe D:\code\xous-core/target/release/ime-plugin-shell.exe D:\code\xous-core/target/release/codec.exe D:\code\xous-core/target/release/modals.exe D:\code\xous-core/target/release/root-keys.exe D:\code\xous-core/target/release/trng.exe D:\code\xous-core/target/release/sha2@0.10.8.exe D:\code\xous-core/target/release/jtag.exe D:\code\xous-core/target/release/status.exe D:\code\xous-core/target/release/shellchat.exe D:\code\xous-core/target/release/pddb.exe D:\code\xous-core/target/release/usb-device-xous.exe D:\code\xous-core/target/release/hello.exe
warning: trait `SerialWrite` is never used
 --> kernel\src\io.rs:5:11
  |
5 | pub trait SerialWrite {
  |           ^^^^^^^^^^^
  |
  = note: `#[warn(dead_code)]` on by default

warning: trait `SerialRead` is never used
  --> kernel\src\io.rs:11:11
   |
11 | pub trait SerialRead {
   |           ^^^^^^^^^^
[... 99 lines cut ...]
INFO:status: checking gateware signature... (services\status\src\main.rs:585)
INFO:status: Guttering Some(Quit) (services\status\src\main.rs:266)
INFO:status: |status: starting main loop (services\status\src\main.rs:703)
INFO:pddb: Requesting login password (services\pddb\src\main.rs:2313)
INFO:pddb: Attempting to mount the PDDB (services\pddb\src\main.rs:2379)
INFO:pddb::backend::hw: decrypted: 8164, FastSpace size: 8164 (services\pddb\src\backend\hw.rs:1318)
INFO:pddb::backend::hw: space_log_addrs len: 1 (services\pddb\src\backend\hw.rs:1344)
INFO:pddb::backend::hw: System BasisRoot record found, generating cache entry (services\pddb\src\backend\hw.rs:1954)
INFO:pddb::backend::basis: Basis .System found and reconstructed (services\pddb\src\backend\basis.rs:1435)
INFO:pddb: PDDB mount operation finished successfully (services\pddb\src\main.rs:2383)
INFO:pddb: _|TT|_PDDB.MOUNTED,_|TE|_ (services\pddb\src\main.rs:719)
INFO:pddb: PDDB post-mount caching stats: 4096 heap, 1144 cache (services\pddb\src\main.rs:772)
INFO:status: audio enable state: false (services\status\src\main.rs:851)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:net::connection_manager: EC rev v0.9.5-0 is incompatible with connection manager (services\net\src\connection_manager.rs:104)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
WARN:llio: Time offsets are not initialized, can't report local time (services\llio\src\lib.rs:44)
```

bunnie, 2024-05-09:
Looks like a regression introduced in Rust 1.78. Steps to reproduce:

Here's the steps to reproduce:

1. update rust to 1.78.0
2. do cargo xtask run
3. expected: see the UI appear in the window. actual: you get a blank screen.

4. do rustup toolchain install 1.77.2
5. run cargo +1.77.2 xtask run
6. expected: same error as in 1.78.0 (if this was a code bug). actual: the UI works just fine

So it seems like the issue is the toolchain, since there was no change to the code between the runs...

It works on Renode and hardware, fwiw. just hosted mode is broken. The hosted mode bug could even be specific to Windows, I don't have a linux host to test this on conveniently at the moment.

bunnie, 2024-05-09:
@xobs any ideas on what could be causing this?

bunnie, 2024-05-09:
The problem seems to be specifically something in between the graphics-server, `minifb`, and the actual windows subsystem.

graphics-server seems to be receiving messages normally, and it can process key hits etc. What seems to be happening is none of the graphics ever get updated to the screen. I tried updating to the latest minifb on crates.io but didn't see any change...

bunnie, 2024-05-09:
Huh, other users have reported this problem on windows 11: https://github.com/emoon/rust_minifb/issues/351

xobs, 2024-05-09:
Does it fix things if you update the `minifb` dependency as they specified?

```toml
minifb = { git = "https://github.com/emoon/rust_minifb.git", rev = "633e6cd4f8f55fd9cbedc461976dde0f408bea35" }
```

bunnie, 2024-05-09:
OK yes, I can confirm that using the gitrev of an unreleased minifb causes the bug to go away:

git = "https://github.com/emoon/rust_minifb.git", rev = "633e6cd4f8f55fd9cbedc461976dde0f408bea35"

I'll swap that pin over to the git repo directly this weekend if no release is made by then.

bunnie, 2024-05-09:
Alright, I have a patch to this, I'll make a PR for it once I'm in a region of the world that's not blocking both ssh and vpn. :grimacing:

bunnie, 2024-05-12:
#539 should improve the situation. If it fixes it for you @rowr111 , please go ahead and close the issue (no hurries tho, just letting you know).

rowr111, 2024-05-14:
yup all good now, thank you!

closed 2024-05-14 by rowr111