betrusted-io/xous-core #541: Jabber / XMPP

sjehuda, 2024-05-19:
Greetings,

I have read [this post](https://www.crowdsupply.com/sutajio-kosagi/precursor/updates/call-for-developers-precursor-chat-client) and I am surprised that there is no mention to XMPP, which is not only a messaging protocol, it is also capable for VoIP, file sharing, publishing (XEP-0060: PubSub), including [visual interface](https://gitgud.io/sjehuda/slixfeed) and anything else.

I have an XMPP account which I use only for XMPP based software (i.e. bots) which serve me with many tasks, including syndication updates, file synchronization, calculator, unit conversion, task scheduler, meet scheduler, and much more.

I suggest to read https://xmpp.org/about/technology-overview/ and make XMPP a priority.

I definitely suggest to focus less on spy-agencies-funded chat protocols.

- https://dpaste.org/tJ3ZH/raw
- https://denshi.org/blog/every-chat-client-absolutely-sucks/every-chat-client-absolutely-sucks/
- https://blog.koehntopp.info/2024/02/13/the-matrix-trashfire.html
- https://gist.github.com/soatok/8aef6f67fec9c702f510ee24d19ef92b

plaurent, 2024-06-20:
The xmpp.rs library does look fairly easy to use. 
https://gitlab.com/xmpp-rs/xmpp-rs/-/blob/main/xmpp/examples/hello_bot.rs?ref_type=heads

But I don’t see any mention of support for end to end encryption. Am I just missing it or is this something that would have to be added?

xobs, 2024-06-21:
`xmpp.rs` looks difficult to use since it uses `tokio` which would require a lot of work to port to Xous

plaurent, 2024-06-21:
Thanks @xobs I was wondering about its dependencies and the implications thereof.

Any suggestions out there?

sjehuda, 2024-06-21:
> Any suggestions out there?

Please join to groupchat "XMPP in Rust" chat (AT) xmpp.rs and I suppose
people will be glad to provide you any reference that you might need.

You might also want to join to groupchat "JDev" jdev (AT) muc.xmpp.org.

Links:
https://search.jabber.network/search?q=xmpp+in+rust
https://search.jabber.network/search?q=jdev