betrusted-io/xous-core #570: Renode emulation broken on `1.15.2.7965 （e6e79aad-202408180425)`

gsora, 2024-08-19:
Renode version: `1.15.2.7965 （e6e79aad-202408180425)`

When running `renode emulation/betrusted.resc`, the log window shows this and emulation stops:

```
emulation/peripherals/vexriscv-aes.cs(18,13): error CS0246: The type or namespace name 'PrivilegeArchitecture' could not be found. Are you missing an assembly reference?
```

I guess this is @xobs realm?

bunnie, 2024-08-19:
I did refactor the AES drivers but it should not impact the target you're hitting - the cramium target now uses rv32 standard extentions instead of the vex custom extensions. so yah, might be a @xobs question...

xobs, 2024-08-20:
This looks to be caused by https://github.com/renode/renode-infrastructure/commit/5256986af3799b0136e91af5d56908cb2fb84bc7 which renamed the type.

I've opened https://github.com/betrusted-io/xous-core/pull/572 to fix this, but it's a breaking change.

PR #572 (2024-08-20): renode: vexriscv: fix CPU on newer Renode versions
Newer versions of Renode rename PrivilegeArchitecture to PrivilegedArchitecture. They appear to have tried to work around this by annotating it with `[NameAlias("privilegeArchitecture")]`, however that doesn't change the fact that the actual type itself changed as well.

This incompatible change will cause breakage on older versions of Renode.

This closes #570
closed 2024-08-20 by bunnie