betrusted-io/xous-core #580: `flatipc` has no `from_ipc_mut()` method

bunnie, 2024-09-22:
The Readme for `flatipc` gives these instructions to recover the original structure from a message:

https://github.com/betrusted-io/xous-core/blob/222f86d709170eff21adaa523420ee35ff6f8435/libs/flatipc/README.md?plain=1#L78-L80

However, I'm not able to find an implementation for `from_ipc_mut`. Or perhaps I'm not including the right trait to find it? I notice the example uses `flatipc_derive:Ipc` as the derive instead of `flatipc::Ipc`, but swapping that out doesn't seem to help.

Can you please clarify the server-side example?

xobs, 2024-09-23:
I'm sorry, you're right, the trait got renamed to `from_slice_mut()` to be more inline with other Rust calls, and I didn't update the README.

Let me update that.

bunnie, 2024-09-23:
I think it'd be also helpful to note that the `signature` field comes from the `offset` field in a memory message. Or at least I think it does? I'm not sure!

xobs, 2024-09-23:
I've updated the documentation in 0.1.2. Additionally, I've added `Ipc::from_memory_message()` that takes care of that for you.

See #581 and https://docs.rs/flatipc/latest/flatipc/