betrusted-io/xous-core #653: i2c read always returns `OK`

samchin, 2025-06-06:
While using the [i2c_lib.rs](https://github.com/betrusted-io/xous-core/blob/69e73ffd61f8f94a40d1ca19062219be6b4eec97/services/cram-hal-service/src/i2c_lib.rs) to create an `i2cdetect` function, I found that reading the bus always returns with an ack even when there is nothing attached to the bus.

bunnie, 2025-06-06:
#654 is the WIP PR to address this issue.

bunnie, 2025-06-06:
It's probably helpful to note that the register interface specification can be found here:

* https://ci.betrusted.io/cramium-cpu/
* https://ci.betrusted.io/cramium/

It's split between two URLs because the code generators for the SoC and the CPU are different.

The page containing docs on the IRQ range is here: https://ci.betrusted.io/cramium-cpu/irqarray7.html

The I2C block comes from the UDMA subsystem of the Pulp SoC open source project. Some more details may be found here but I have struggled to find a definitive reference from Pulp project that matches the latest RTL: https://raw.githubusercontent.com/pulp-platform/pulp/master/doc/datasheet.pdf

The other helpful resource when extending low level drivers is the actual UTRA header file for all the hardware blocks, which can be found here: https://github.com/betrusted-io/xous-core/blob/main/utralib/src/generated/cramium_soc.rs

bunnie, 2025-06-06:
I haven't tested the code yet but #654 is broken into individual commits that try to walk through the process of adding an API call to a service, it might be helpful to look at the commits if you're interested in how this all works under the hood.

bunnie, 2025-06-07:
#654 is now ready for review. I ended up deciding to still have i2c_read return "OK", but just change the type of OK returned. So it can either return Ok(I2cResult::Ack(length)) or Ok(I2cResult::Nack).

Err(_) types are reserved for unrecoverable OS errors, e.g. messages failed to be delivered because e.g. the HAL service crashed.

closed 2025-06-09 by samchin